realthunder/FreeCAD_assembly3 #302: Detect broken Subshape Binder Supports

ceremcem, 2020-06-18:
I have experienced exactly [this broken support issue](https://github.com/realthunder/FreeCAD_assembly3/issues/273#issuecomment-523007087). 

This is a very serious issue that always go unnoticed until the actual production. Moreover it is not that rare, it seems that it can happen any time.

I modified your example script in https://github.com/realthunder/FreeCAD_assembly3/issues/89#issuecomment-424979787 which checks every binder's support and reports the broken ones. I think it should be hooked in the `Refresh` step, the binders stay verified every time. Please consider adding it to the refresh hook.

```py
import FreeCAD

# TODO: 
# Should be recursive: It should be run on dependencies too

echo = FreeCAD.Console.PrintMessage
warn = FreeCAD.Console.PrintWarning
error = FreeCAD.Console.PrintError
doc = FreeCAD.ActiveDocument

# TODO: Refactor this:
# "exclude" means "processed"

def find_path(parent,tp,exclude):
    ret = []
    for sub in parent.getLinkedObject(True).getSubObjects(1):
        obj = parent.getSubObject(sub,retType=1)
        if not obj or obj in exclude:
            continue
        if obj.isDerivedFrom(tp):
            ret.append(sub)
        else:
            for ssub in find_path(obj,tp,exclude):
                ret.append(sub+ssub)
        exclude.add(obj)
    return ret

def get_bound_object(binder):
  # found in FreeCAD/src/Gui/propertyeditor/PropertyItem.cpp in function LinkSelection::select 
  return binder.Support[0][0].getSubObject(binder.Support[0][1][0])

def find_objs(tp='PartDesign::SubShapeBinder', doc=App.ActiveDocument):
    from collections import defaultdict
    ret = defaultdict(list)
    objs = doc.Objects
    exclude = set()
    for obj in objs:
        # only checking group type objects
        if not obj.hasChildElement() and \
           not obj.hasExtension('App::GeoFeatureGroupExtension'):
            if obj.Name.startswith('Binder'):
              if not get_bound_object(obj):
                #path = '.'.join(map(lambda x: x.Label, obj.InListRecursive[:2])[::-1])
                error("* Broken binder: %s in %s\n" % (obj.Label, doc.Name))	  
                Gui.Selection.addSelection(doc.Name, obj.Label)
            continue
        if obj in exclude:
            continue
        subs = find_path(obj,tp,exclude)
        if subs:
            ret[obj] += subs

    # filter out non-root-level groups
    for obj in exclude:
        ret.pop(obj,None)

    # find root-level links to those root-level groups. Omit this step if you
    # don't want to include root-level links
    filtered = [ o for o in objs if o not in exclude and o not in ret ]
    for obj in filtered:
        subs = ret.get(obj.getLinkedObject(True),None)
        if subs:
            ret[obj] = subs
    return ret

def check_supports(tp='PartDesign::SubShapeBinder'):
  Gui.Selection.clearSelection()
  for name, doc in App.listDocuments().items():
    echo("* Checking document %s\n" % name)
    find_objs(tp, doc)

echo("\n\n\n")
echo("----------------- Checking SubShape Binders'  Supports -----------------\n")
check_supports()
echo("done.\n")
```

realthunder, 2020-06-18:
I have fixed SubShapeBinder to throw exception in case it cannot get the binding object. There is will be checking on document restore. So you'll notice the error when you load the document.

closed 2020-06-18 by realthunder

ceremcem, 2020-06-18:
Works great! Thanks.