realthunder/FreeCAD_assembly3 #305: Ctrl+x shortcut does not trigger the same event with right click -> cut

ceremcem, 2020-07-03:
In spreadsheet, `Ctrl+x` does not create the red dashed border as the context menu -> cut does. 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23691 (Git)
Build type: Release
Branch: LinkDev
Hash: 99d4ed1196b87749e27dc3481cc8fa940bff3dc3
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2020-07-03:
I can't reproduce this with either Python2 or 3, Qt4 or Qt5, native or lxc container. Maybe this is your system's UI problem?

ceremcem, 2020-07-03:
Sorry, me neither. It might be my mistake. It currently works.

closed 2020-07-03 by ceremcem