realthunder/FreeCAD_assembly3 #306: Avoid unnecessary recomputes on spreadsheet changes

ceremcem, 2020-07-03:
Currently any spreadsheet change triggers a recompute, which kind of freezes the whole app for a bunch of seconds. 

I think recomputation should not be triggered on: 

* Cut + paste
* Insert a string (non-variable thing)
* Add a row 
* Remove a row which doesn't contain any variable 

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23691 (Git)
Build type: Release
Branch: LinkDev
Hash: 99d4ed1196b87749e27dc3481cc8fa940bff3dc3
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2020-07-04:
I did a rebase recently with an important fix. Make sure you sync to the latest revision.

closed 2020-07-04 by realthunder

ceremcem, 2020-07-05:
Adding new row still triggers the recomputation, and also prints the following errors to the ReportView: 

```
<Gui> ViewProviderDocumentObject.cpp(743): no view
<Gui> ViewProviderDocumentObject.cpp(743): no view
<Gui> ViewProviderDocumentObject.cpp(743): no view
```

# Version 

```OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23711 (Git)
Build type: Release
Branch: LinkDev
Hash: 1418e7f006ecb435e763601b7c50d36d826a651e
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2020-07-15:
I just added a commit to fix this.