realthunder/FreeCAD_assembly3 #307: Is this transient exception important?

ceremcem, 2020-07-10:
In the recent version (I don't know where it actually starts) I noticed that changing one of the parameters cause an exception, but the exception disappears after a prior automatic refresh cycle and resulting model seems to be valid. 

Try changing `driver_base.param/bend_inner_radius`. Here is the exception: 

```
<Gui> ViewProviderDocumentObject.cpp(744): no view
<Gui> ViewProviderDocumentObject.cpp(744): no view
<Gui> ViewProviderDocumentObject.cpp(744): no view
<Gui> ViewProviderDocumentObject.cpp(744): no view
<App> PropertyLinks.cpp(438): driver_base#_Element263.LinkedObject missing element reference driver_base#Fusion006 ;#5707;:M#7e67;RFI;:T6165:5:F.Face43
Traceback (most recent call last):
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 695, in execute
    info = self.getInfo(False)
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 1118, in getInfo
    self.getElementSubname(),False,noShape)
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 1594, in getElementInfo
    '{}.{}'.format(objName(part),subname))
<type 'exceptions.RuntimeError'>: Failed to get geometry element from driver_base#Fusion006.?Face43
Recompute failed! Please check report view.
156.703522 <asm3.sys> sys_slvs.py(63): redundant constraints
0.000117 <asm3.sys> sys_slvs.py(68): dof remaining: 0
1.076333 <asm3.sys> sys_slvs.py(68): dof remaining: 0
0.711645 <asm3.sys> sys_slvs.py(68): dof remaining: 0
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23721 (Git)
Build type: Release
Branch: LinkDev
Hash: 61f836ab4165f7c83e04d44a054ccaf10349116f
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

20200710T1738-revision-14.zip sent via Forum DM.

realthunder, 2020-07-11:
Do you remember which parameter you changed?

realthunder, 2020-07-11:
BTW, I just found some `unnatural` linking in your document. `Link010` in `button_driver` links to an assembly `material` in `driver_base`. However, all spreadsheets inside `material` actually lives in `button_driver`. Any reason for this 'circular' linking? It looks like a simple group inside `button_driver` will do?

ceremcem, 2020-07-11:
> Do you remember which parameter you changed?

It's the `driver_base.params/bend_inner_radius` parameter. Try slight changes, like `1.5mm` to `1.6mm`, etc. 

> Any reason for this 'circular' linking?

No particular reason. My intention was creating an assembly that only holds those "material definition spreadsheet"s and then create link in any higher level model (`button_driver` in this example). 

Those material definition sheets are used (searched by their labels, found and their content is used) during an unfold operation. 

[image]

ıt looks like it is completely how it should be. The original document seems to live in `driver_base` and in the `button_driver` there is a link to that assembly. 

Am I misinterpreting the icon images?

realthunder, 2020-07-11:
> Am I misinterpreting the icon images?

Yeah, you are right. Nothing wrong here. I just hit a bug here. I'll check the exception tomorrow.

realthunder, 2020-07-12:
I did find something wrong. But not sure if it is the direct cause. Please sync my branch and try.

BTW, I have added a new feature for easy hierarchy navigation. The command is called `Std_SelUp`. The default keyboard shortcut is 'U, U'. You can type the shortcut while hover your mouse over any tree item, or any geometry in 3D view. A menu will pop up showing the relevant hierarchy. If there is no object under the mouse, then the last browsed hierarchy will be shown. Supported operations on the menu items are,

* Left Click to select the corresponding hierarchy.
* Right Click to show children of that hierarchy.
* Shift + Left Click to edit the object (equivalent to double click the item in tree view). 
* Shift + Right Click to show edit menu of that object.

This `SelUp` is intended to be an extra dimension of the tree view. It can replace the tree view  in most use cases, and offers a more efficient way of hierarchy navigation

ceremcem, 2020-07-12:
Still gives the same(?) exception: 

```
<Gui> ViewProviderDocumentObject.cpp(744): no view
<Gui> ViewProviderDocumentObject.cpp(744): no view
<Gui> ViewProviderDocumentObject.cpp(744): no view
<App> PropertyLinks.cpp(444): driver_base#_Element263.LinkedObject missing element reference driver_base#Fusion006 ;#5707;:M#7e67;RFI;:T6165:5:F.Face43
Traceback (most recent call last):
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 695, in execute
    info = self.getInfo(False)
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 1118, in getInfo
    self.getElementSubname(),False,noShape)
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 1594, in getElementInfo
    '{}.{}'.format(objName(part),subname))
<type 'exceptions.RuntimeError'>: Failed to get geometry element from driver_base#Fusion006.;#5707;:M#7e67;RFI;:T6165:5:F.Face43
Recompute failed! Please check report view.
119.458669 <asm3.sys> sys_slvs.py(63): redundant constraints
0.000106 <asm3.sys> sys_slvs.py(68): dof remaining: 0
1.122018 <asm3.sys> sys_slvs.py(68): dof remaining: 0
0.750176 <asm3.sys> sys_slvs.py(68): dof remaining: 0
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24353 (Git)
Build type: Release
Branch: LinkDev
Hash: f260b6b0fdfd0eb95a9919de3d2e9b4c342891e4
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2020-07-12:
> BTW, I have added a new feature for easy hierarchy navigation. The command is called Std_SelUp

OMG, that's fantastic! Thank you!

ceremcem, 2021-02-14:
Closing as this is an old issue which can not be reproduced.

closed 2021-02-14 by ceremcem