realthunder/FreeCAD_assembly3 #308: the sketches disappear (in the sense of display) from the pattern surface

tiket18, 2020-07-14:
Let's create a solid body. Place a sketch on its surface = you can see the excise on the damage (this is necessary for use in the Path workbench). Then we will perform an operation with this solid body (for example) to combine it with another body. Or we'll make a mirror copy. And there is no sketch image on the surface. We have to create a link to the sketch and move it separately (for overlay).
[image]

realthunder, 2020-07-16:
Can you please provide a file demonstrate the problem? I don't quite get what you mean here. Do you mean that the sketch some how is obscured by the solid? In my branch you have this on top when selected feature. Where the selected object is shown regardless of its visibility property or any obstructing object. If you want the object stay visible on top even without selection. Select it first, and press CTRL+SHIFT+SPACE. The object will stay showing on top of any other objects. Press the shortcut again to toggle it.

tiket18, 2020-07-16:
Here is an example - a body is created, on its surface there is a sketch (in the "Model" tree the sketch is inside the body). Then this body => create a link and (e.g.) merge with another body = the sketch becomes invisible (in the picture to the right the body is the original and to the left the same body but merged with another body).
And just "display on top of everything" is not quite what you need - I don't need to look, I need to use the Path workbench, where to set the path, you need the tracker to be included in the Model. And in the Model, these trajectories (according to the sketch) are visible or not visible - they disappear and appear, I can't understand the system.
[image]

tiket18, 2020-07-16:
I put different parts (bodies) into one (body, union) to be placed on the desktop of the machine for subsequent turning (machining on both sides).

tiket18, 2020-07-16:
Here you go. After Fusion was created, the sketches from the surface disappeared.
[image]
Here you go. After Fusion was created, the sketches from the surface disappeared.
FC file does not allow to attach, if inform your e-mail = send.

DevyHeavy, 2020-07-17:
@tiket18, I think I understand what you are saying.

You want the lines and geometry in your sketches to appear on your solid.

One way to do this is to use the Part Design workbench. First, select the dropdown menu and then the Part Design workbench, next go to Edit > Preferences > Part Design. In the preferences in the "Model settings" uncheck everything. Now if you make a closed or valid (all green) sketch and extrude it into your solid the lines will stay. This is how I make my scribe lines for parts.

This doesn't seem to be an Assembly3 issue (as far as I can tell from your questions).

realthunder, 2020-07-17:
@tiket18, if this is really what you mean as DevyHeavy mentioned, then the following two options are what caused your problem. Just uncheck them. You need to activate PartDesign workbench first and then open the preference dialog to see the PartDesign options.

[image]

tiket18, 2020-07-17:
1. I agree, of course, this is not a problem of this particular assembly, but of the entire program.
2. No, it's not about that (using the sketch). I create a body and squeeze it out. And at the same time I create an additional sketch on the surface - for carving, I do not squeeze it out and do not make a groove, I then set the path on the edge - a sketch is needed for this.

[image]

tiket18, 2020-07-17:
[image]
Changing the check does not change the view - sketches should be visible on all figures.

realthunder, 2020-07-17:
Just rename your document to a .zip file, and then you can attach the file here.

tiket18, 2020-07-17:
hesitates
[image]

tiket18, 2020-07-17:
It doesn't. Even if not renamed, but honestly archived.

realthunder, 2020-07-17:
That's strange. You can email me at realthunder.dev at gmail

tiket18, 2020-07-17:
Sent. Please check. The Good Corporation likes to spam everything.

realthunder, 2020-07-17:
I got the file. What you want cannot be easily achievable. And you are mixing Part And PartDesign features which can easily cause more confusion. Following DevyHeavy's idea, you can create a reverse pad into each body using the sketch to create the edge for carving. Make sure you have unchecked those two 'Refine' options above. The attached file is a renamed zip.

[sketch_pad.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4937323/sketch_pad.zip)

luzpaz, 2021-07-13:
Closing ticket as there's no response from @tiket18

closed 2021-07-13 by luzpaz