realthunder/FreeCAD_assembly3 #309: Crash on second Sketcher rectangle tool placement

DevyHeavy, 2020-07-16:
**BUILD INFO**
```
OS: Ubuntu 20.04 LTS (XFCE/xubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2445 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: b06c7d8f094fe3631d8741d19a0fb1fca49e5742
Python version: 3.7.6
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

When using the latest Assembly3 appimage, if you are using the sketcher rectangle [tool](https://wiki.freecadweb.org/Sketcher_CreateRectangle) twice in an instance of FreeCAD then the program will crash.

It does not matter if you place a rectangle and then close out of the sketch before placing the next rectangle, it will still crash.

Let me know about any questions, thanks.

realthunder, 2020-07-17:
This is a serious bug. Thanks for the report. I'll make a release soon.

closed 2020-07-17 by realthunder

realthunder, 2020-07-17:
Image updated.

DevyHeavy, 2020-07-17:
Thanks for your quick response and work.

realthunder, 2020-07-17:
Woops, just found a UI problem. Re-uploaded image just now. If you're using AppImage, you can use AppImageUpdate for quick update.