realthunder/FreeCAD_assembly3 #310: Sketcher: constraints related to deleted elements are not removed.

DevyHeavy, 2020-07-17:
**BUILD INFO:**
```
OS: Ubuntu 20.04 LTS (XFCE/xubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2451 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 9cde54a5bb433978a0f724a0879aad1f3c91a0ca
Python version: 3.7.6
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```
When using sketcher, if an element with an (any?) constraint is deleted, the related constraint is not removed from the sketch.

For example, vertical and horizontal constraints are supposed to be removed when the related line is deleted, but this does not happen in the latest image.

TEST: Open Part Design workbench, create a sketch, draw a vertical line with vertical constraint attached, and delete line or press control+z to get rid of the line.

TEST RESULT: Line removed but constraint still exists in the sketch.

realthunder, 2020-07-17:
Looks like the sketch is having quite some problems. I recently did a merge with the upstream, and there were indeed a few conflicts in sketch code base. I'll fix it soon.

DevyHeavy, 2020-07-17:
OK. If it is alright with you, I'll continue making bug reports for problems I find. Maybe that will help verify that sketcher is in good shape later.

realthunder, 2020-07-17:
Yes, that'll be the best. Thanks

closed 2020-07-17 by realthunder

realthunder, 2020-07-17:
Just released another update. Please check.