realthunder/FreeCAD_assembly3 #312: the dependencies are broken when the value in the table that does not affect the sketch is changed

tiket18, 2020-07-26:
[image]

There is a set of sketches, there is a table with values.
In general, it is noted that the solver of dependencies in the sketch can arbitrarily (on a flat place) take and count dependencies and boldly declare them unsolvable or conflicting. And so I was able to achieve stable reproduction of the problem itself will send the file by email (please check spam).

The algorithm is...
1. open the file and see the body + two Abort1 & abort2 sketches - you can open them, etc., the solver does not see any conflicts. These thumbnails have references to other thumbnails that use data from the table. Among other sketches, B4 is used in sketch 1 only to determine hole diameters, which are not referenced anywhere.
2. change the B4 value in the table to 7 = sketches abort1 & abort2 are broken.

Additionally another problem (can also be reproduced) - if you delete thumbnails 61 and 71 = body 2 is destroyed in the just opened sent file - objects 3, 4 and 5 leave body 2.

realthunder, 2020-07-27:
Both abort1 and abort2 has multiple external geometry reference to sketch 1. This is why they are recomputed. As to why the sketches are broken, I guess there are some problem in the sketch solver. When did you author this file? Maybe it is due to recent changes of sketch solver in upstream. I did a merge with upstream in 07.15 release.

BTW, you can use `Tools -> Dependency graph` to get a better view of the dependency.

[image]

tiket18, 2020-07-27:
I created the file before July 15th. Such breakdowns often happen - I can create a curve in a sketch and set the Drop-down points, and the sketch breaks down and the solver swears at many restrictions, completely unrelated to this point.

I created the file before July 15th. Such breakdowns often happen - I can create a curve in a sketch and set the Drop-down points, and the sketch breaks down and the solver swears at many restrictions, completely unrelated to this point.
Specifically from this sketch - I change the value of one diameter, from which in those sketches that break, absolutently nothing depends. In a sketch which uses this diameter value (1), only the diameter changes and nothing moves for the purpose of sketches abort.

tiket18, 2020-07-27:
More than that. We open the file, we don't change anything in it at all, no values. Open the thumbnail abort1 and expand the sub-menu Advanced solver settings, do not change anything and click Resolve - externally no negative changes. But if you just close the sketch = breakage.

luzpaz, 2022-07-04:
@tiket18 can you revisit this ticket ?

closed 2022-07-04 by tiket18