realthunder/FreeCAD_assembly3 #315: Addon Manager not working

FredericLeuba, 2020-09-07:
Hello,

The addon manager "Workbench" tab is always empty.
I added the url to the official github FreeCADAddOns repo, but it does not fix anything.
Clicking on the "Macro" tab triggers a fetch and shows a list of macros.

Can this be fixed?

Best regards
Fred

realthunder, 2020-09-07:
It is working on my computer. Can you please paste the FreeCAD About information here? Please make sure you are using the latest release.

FredericLeuba, 2020-09-17:
Hello Realthunder,

first I wish to warmly thank you for the incredible job you do, and the benefit individuals and micro-companies like mine get from it. Don't hesitate to tell me if I can help, and how.

Now, here is the FreeCAD config 

OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git)
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.7.8
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: French/France (fr_FR)

Regards
Fred

realthunder, 2020-09-17:
You are running the latest image. The addon manager is working on my MacOS. Any error message in report view? Did you config any proxy settings in addon manager?

You can also try the upstream FreeCAD release. Please let me know if the addon works there. If so, you can install the addon there, and it will also be available when using my image. If you can't use the addon manager there either, I suggest you report the problem to FreeCAD forum.

madpilot, 2020-09-19:
I'm also seeing this issue

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git)
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.7.8
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/Australia (en_AU)
```

Reverting to upstream works fine.

snelweg, 2020-10-14:
Exactly the same problem here, empty workbenches
[image]

`OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.10.13.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at FETCH_HEAD)
Hash: bb9ae8c4442badbfeb5ad4e13908472287604c7d
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
`

realthunder, 2020-10-14:
Probably not the same problem as OP, because I also see this problem now, and so does using upstream release. I think it is because github changed its page content. @luzpaz can you confirm?

luzpaz, 2020-10-14:
@realthunder I can't reproduce.  
```
OS: Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22670 (Git) AppImage
Build type: Release
Branch: master
Hash: 12155f4aab09047c5697db0c1b3cf93b02edda03
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

luzpaz, 2020-10-14:
@realthunder correction: I **can** reproduce on Linkstage3 branch (not on master)
```
OS: Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.7.8
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

snelweg, 2020-10-14:
> @realthunder I can't reproduce.
> Version: 0.19.22670 (Git) AppImage

It happens in the Linkstage release

> Version: 2020.10.13.14555 (Git shallow) AppImage

realthunder, 2020-10-14:
Right, just checked the latest 22670, and it is working. I tried previously 22474 and has the same problem. Must have been fixed recently. I'll do a sync with the master soon.

snelweg, 2020-10-14:
> Right, just checked the latest 22670, and it is working. I tried previously 22474 and has the same problem. Must have been fixed recently. I'll do a sync with the master soon.

Thanks very much, you are the best!
Been using FreeCAD off and on for a few years now. I switched over for good since I found your branch. Keep up the good work, it is very appreciated!

luzpaz, 2020-10-14:
@realthunder yes, it was fixed somewhere between those commits. Do you need the exact commit ?

snelweg, 2020-10-14:
https://forum.freecadweb.org/viewtopic.php?f=3&t=51150

snelweg, 2020-10-16:
> New daily builds are available now.

Great!!!
It all works again now. Thanks for all the little treasures you made available in FreeCAD now. So nice to use it now. Very happy and will spread the word.

luzpaz, 2020-10-27:
Ticket can be closed

MeanEYE, 2020-12-14:
Just downloaded daily and it's not working.

[image]

OS: Debian GNU/Linux bullseye/sid (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.7.8
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2020-12-15:
@MeanEYE Please try with the latest `Daily` AppImage.

MeanEYE, 2020-12-15:
Before posting, I downloaded latest AppImage available under releases page. I'll try again today with newer one.

snelweg, 2020-12-15:
> Before posting, I downloaded latest AppImage available under releases page. I'll try again today with newer one.

Are you sure?

https://github.com/realthunder/FreeCAD_assembly3/releases

MeanEYE, 2020-12-15:
You are right @snelweg. I didn't download daily, even though it was under release titled "Daily", but file name didn't contain `Daily` in it. Am re-downloading and will test again.

MeanEYE, 2020-12-15:
Downloaded version:

MeanEYE, 2020-12-15:
Downloaded version:

OS: Debian GNU/Linux bullseye/sid (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.14.23076 +3015 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: eb10f4011b36086ed70e8f667c70c10c6d2ecb3e
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

And output is still:
[image]

realthunder, 2020-12-16:
Are there any error message in the Report View?

MeanEYE, 2020-12-16:
Had to open it manually but it has this:
```
12:22:02  Debug: addon_flags.json not found
```

realthunder, 2020-12-16:
Can you manually download this file (https://raw.githubusercontent.com/FreeCAD/FreeCAD-addons/master/addonflags.json)? If not, maybe there is something wrong with your network? Like ssl certificate problem?

Can you also try with upstream AppImage?

MeanEYE, 2020-12-16:
I can easily download that file, however it's important to note that in debug message file has underscore in it. There's nothing wrong with my network. File is hosted on GitHub's user content domain. If my network had issues resolving that domain none of the images on GitHub would work either, which is not the case. 

Where can I find this upstream image?

realthunder, 2020-12-16:
According to the addon manager code, the debug print (which got the name wrong in the message) will only print if the above url (which is correct) failed to download.

You can get the AppImage from https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre

MeanEYE, 2020-12-16:
Okay, this image worked!

realthunder, 2020-12-17:
Just made a new release [12.17](https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11).

luzpaz, 2020-12-17:
@realthunder is the SSL issue affect upstream 0.19 as well ?

realthunder, 2020-12-17:
Nope. I just found out that this problem was discussed [here](https://forum.freecadweb.org/viewtopic.php?f=4&t=34873&start=20#p327416).

fredolaredo, 2020-12-28:
Hi,
Read all the discussion , upgraded to 12.17 version, reinit preferences from scratch (removed my ~/Library/FreeCAD folder), even cloned the add-on git repo manually to the Mod folder and renamed the "addonflag.json" file to "add-on_flag.json" but no luck so far.

OS: macOS High Sierra (10.13)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.17.23076 +3024 (Git)
Build type: Release
Branch: LinkStage3
Hash: 560eb2c4645cd548b6f718c645c7af70384ff54c
Python version: 3.7.9
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2020-12-29:
Please try again with the latest release 12.29

fredolaredo, 2020-12-29:
looks like it is working fine now, except for the following message 

```
16:12:31  A3 available
16:12:36  During initialization the error "invalid literal for int() with base 10: '12.29'" occurred in freecad.cadenas3dfindit
16:12:36  --------------------------------------------------------------------------------
16:12:36  Traceback (most recent call last):
  File "<string>", line 156, in InitApplications
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/fred/Library/Preferences/FreeCAD/Mod/3DfindIT/freecad/cadenas3dfindit/init_gui.py", line 146, in <module>
    freeCadVersion = int(FreeCAD.Version()[1])
ValueError: invalid literal for int() with base 10: '12.29'
16:12:36  --------------------------------------------------------------------------------
```

luzpaz, 2021-02-22:
@fredolaredo Looks like the issue is related to the 3DfindIT addon and not to the Addon Manager itself.  Can you update the addon, restart and see if the error is still there?

fredolaredo, 2021-06-05:
sorry for the delay , I didn't see your comment.
Addon Manager works fine since then even with 3DFindIt

luzpaz, 2021-06-05:
Great. Then closing this ticket.

closed 2021-06-05 by luzpaz

andyylin, 2022-05-03:
Hi, I'm also having this problem with the latest Daily release. I've checked my firewall and it should be working fine.

```
OS: macOS 12.3
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: C/Default (C)
```

Here are the Report View logs:
```
Traceback (most recent call last):
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1040, in _send_output
    self.send(msg)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 980, in send
    self.connect()
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1454, in connect
    self.sock = self._context.wrap_socket(self.sock,
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/ssl.py", line 1040, in _create
    self.do_handshake()
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
15:05:10  ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)
15:05:10  
During handling of the above exception, another exception occurred:

15:05:10  Traceback (most recent call last):
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/Mod/AddonManager/addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1349, in do_open
    raise URLError(err)
15:05:10  urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>
15:05:10  Debug: addon_flags.json not found
15:05:10  Traceback (most recent call last):
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
15:05:10    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1040, in _send_output
[... 50 lines cut ...]
    self.do_handshake()
15:05:14    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
15:05:14  ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)
15:05:14  
During handling of the above exception, another exception occurred:

15:05:14  Traceback (most recent call last):
15:05:14    File "/Applications/FreeCAD realthunder.app/Contents/Mod/AddonManager/addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
15:05:14    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
[... cut ...]
15:05:14    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
15:05:14    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1349, in do_open
    raise URLError(err)
15:05:14  urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>
15:05:14  Appears to be an issue connecting to the Wiki, therefore cannot retrieve Wiki macro list at this time
```