realthunder/FreeCAD_assembly3 #316: Perhaps considering a grant from the Shuttleworth Foundation?

luzpaz, 2020-09-11:
@realthunder what do you think about trying to apply for this:
https://shuttleworthfoundation.org/thinking/2020/09/09/thinking-application-webinars/
It may be a way to help fund your efforts on top of donations from the community?

realthunder, 2020-09-12:
Thanks. I'll check out.

qingfengxia, 2020-09-15:
I have some other funding source, I have contacted you via forum message.  or quick by my      qingfeng.xia            HOtmail

luzpaz, 2021-02-21:
@realthunder :point_up_2:

timmwille, 2021-02-26:
Thank you for raising this topic, I'm currently myself on the topic how to finantially support FreeCAD. I have an Option that might be working for sustsinably funding the development for the next 2 years.
I'm currently asking myself if it makes sense to create a FreeCAD Foundation for this purpose? or just pay the development individually? or use Open Source Ecology or another Organisation as vehicle to provide the funds? Where would I address this topic best? Maybe @cenemem you could help me find the people to straightforward way of getting feedback on this? Does it make sense in the forum even?

luzpaz, 2021-02-26:
You could connect @yorikvanhavre and @kkremitzki on the FreeCAD forum about this.

timmwille, 2021-03-02:
I've placed it here, I hope this is as precise as possible to the point
https://forum.freecadweb.org/viewtopic.php?f=8&t=12242&p=483581#p483581