realthunder/FreeCAD_assembly3 #317: GMSH During Install

Velchao, 2020-09-11:
Hello,

Thank you for your great work on FreeCAD.

I was hoping to know if there is a reason why in V0.19, Revision 21911 + 2532 (Git), 2020/08/11 11:08:12, that GMSH is not installed as part of the images available for download https://github.com/realthunder/FreeCAD_assembly3/releases.

It is my understanding that the standard FreeCAD release installs with it a consistent version of GMSH with it.

Many thanks in advance!

realthunder, 2020-09-12:
Which workbench uses GMSH, and what command? Is it MeshPart?

Velchao, 2020-09-12:
The workbench GMSH uses is called, “FEM”. FEM meshes are not the same as those produced from the “Mesh Design” workbench. The gmsh binary path is set in the FEM workbench preferences, but the installation files do not include gmsh.exe.

In v0.18, my directory, for example, is: C:/Program Files/Engineering Software/FreeCAD 0.18/bin/gmsh.exe

realthunder, 2020-09-12:
I see. Can you please copy the gmsh.exe file from upstream (like the 0.18 directory you mentioned) to my branch directory, and see if it works?

Velchao, 2020-09-14:
The .exe file itself is 55.3 MB, exceeding the ability to simply upload on GitHub.

I have looked through the main FreeCAD git repository, https://github.com/FreeCAD/FreeCAD , and I cannot find the .exe file there, either.

The gmsh source code is here, though: https://gitlab.onelab.info/gmsh/gmsh.

Please let me know if there is something I am missing.

realthunder, 2020-09-26:
I just found out that gmsh exists as a Python pip package. Can you try install it by typing the following command in FreeCAD Python console and see if it works for you. If so, I'll include it in the next release.

```python
import subprocess, sys, os
subprocess.check_call([os.path(sys.executable)+'\\python.exe', '-m', 'pip', 'install', 'gmsh'])
```

luzpaz, 2021-02-03:
@Velchao ping

luzpaz, 2021-02-20:
@realthunder close this ticket?

closed 2021-06-24 by luzpaz