realthunder/FreeCAD_assembly3 #318: unable to view assembly strucure/components during step import

anirudhnehra, 2020-09-19:
Hi,
I cant seem to see all the parts that make up step file during its import. What to do?

Thanks
[image]

realthunder, 2020-09-19:
First, make sure you switch to `Part` workbench at least once. Then go to Preference -> Import/Export, and do the following settings. The most important one is to **NOT** enable STEP compound merge. Also note that, not all STEP contains assembly.

[image]

anirudhnehra, 2020-09-20:
Thank you, Working good after unchecking compound merge option.

Regards 
Anirudh

closed 2020-09-20 by anirudhnehra