realthunder/FreeCAD_assembly3 #319: appimage has not been updated for more than a month.

tiket18, 2020-09-23:
During this time, something useful always happens in the main branch. Please make a new build further and preferably at least once every two weeks. I'm especially worried about the path workbench, which is developed only in the main branch.

FlailAway, 2020-10-09:
Seriously? Such ingratitude, not even thanking the Dev for spending the 1000s of hours spent building this Branch.

Why not learn C++ and Python and help with it.

Just when I thought I had seen it all in life, I see this shameful request.

tiket18, 2020-10-11:
Не могу себе представить, что в моем сообщении могло быть воспринято как повод для обиды. Единственная версия - дефект перевода, языковой барьер. Увы, я использую depl.com.
I can't imagine that my message could have been taken as a reason for offence. The only version is a translation defect, a language barrier. Alas, I use depl.com.

ceremcem, 2021-02-21:
@tiket18 There is nothing wrong with your comment. Anyone might also get frustrated because of a problem while trying to get a job done, that's even okay.

ceremcem, 2021-02-21:
Regarding to the title, this issue is resolved and should be closed.

closed 2021-02-21 by tiket18