realthunder/FreeCAD_assembly3 #320: External Reference or ShapeBinder External Geometry

DaAwesomeP, 2020-09-26:
Hello!

It does not yet seem possible to transfer, reference, or bind geometry from parts in different files onto each other.

One use-case for this is an object with mounting holes. You might constrain the object to fit next to your part, and then you want to cut out matching holes from your part.

Here is an example:
[image]

I have tried:

 - Shapebinding a sketch
 - Creating an external reference circle in the part sketch
 - Copying the sketch with the mounting holes

Or maybe there is a way to do this and I don't know what it is. The holes don't necessarily need to be permanently linked/referenced, so any way to copy the geometry would work in this situation.

Relevant forum post: https://forum.freecadweb.org/viewtopic.php?f=20&t=50541

ceremcem, 2021-02-21:
Waiting user response due to https://forum.freecadweb.org/viewtopic.php?f=20&t=50541#p434444

luzpaz, 2021-07-30:
@DaAwesomeP ping

luzpaz, 2022-07-11:
Closing ticket as no response from @DaAwesomeP

closed 2022-07-11 by luzpaz