realthunder/FreeCAD_assembly3 #322: Shadow & transparency not appearing correclty

jaisekjames, 2020-10-14:
My shadow look like this
[image]

My dock window become transparent. But object also become transparent. See three cubes are there. If select cube, it will appear [same as top]. 
[image]

Freecad:
OS: Manjaro Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.10.13.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at FETCH_HEAD)
Hash: bb9ae8c4442badbfeb5ad4e13908472287604c7d
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/India (en_IN)

Graphics - ATI RV515

realthunder, 2020-10-14:
It's most likely a graphics driver issue. Any chance to update your graphics driver? Also, please try spot light in shadow. Is everything fine outside of shadow display mode?

jaisekjames, 2020-10-16:
Thank for quick responds. may be graphics driver issue. Graphics driver come with default linux driver set. It is an old card  ATI Mobility Radeon X2300 [Compaq 6910p]. Is there any OpenGL min version requirement for this to work?

realthunder, 2020-10-16:
Not sure about the min version. I did get some complains about ATI cards. Maybe you can try update your driver and see.

luzpaz, 2021-02-22:
@jaisekjames any progress ?

jaisekjames, 2021-02-22:
I am using manjaro Linux with gnome. & It is not possible to update driver since only available driver is come with Linux default one. I will close this issue.

closed 2021-02-22 by jaisekjames

luzpaz, 2021-02-22:
bummer