realthunder/FreeCAD_assembly3 #324: Inconsistent constraints error is well... inconsistent

akamensky, 2020-10-19:
I am learning assembly 3, so if this is something me doing wrong -- feel free to close.

I've created 2 bodies (externally imported STEP) and dropped them into "Parts" of assembly. I select faces that should be aligned and click `Create "PlaneAlignment" constraint`. The I click `Solve constraints` and I immediately get:

```
22:32:32  159.844357 <asm3.sys> solver.py(132): List of failed constraint:
___Bed_assembly#Constraint, handle: 2
22:32:32  159.846334 <asm3.gui> gui.py(452): command "asm3CmdSolve" exception
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.8/site-packages/freecad/asm3/solver.py", line 109, in __init__
    self.system.solve(group=self.group,reportFailed=reportFailed)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.8/site-packages/freecad/asm3/sys_slvs.py", line 67, in solve
    raise RuntimeError(reason)
RuntimeError: inconsistent constraints

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 611, in report
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.8/site-packages/freecad/asm3/solver.py", line 416, in solve
    return _solve(*args,**kargs)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.8/site-packages/freecad/asm3/solver.py", line 386, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.8/site-packages/freecad/asm3/solver.py", line 112, in __init__
    raise RuntimeError('Failed to solve {}: {}'.format(
RuntimeError: Failed to solve ___Bed_assembly#Assembly: inconsistent constraints
```

Clicking `solve constraints` again may or may not work. If it does not work I keep getting same error, once it worked I don't get error again (without any other changes, just keep clicking and it works "eventually").

realthunder, 2020-10-19:
Can you please post the file you've tried?

realthunder, 2020-10-20:
I replied on Discord. Just for completeness, I copied it here.

You didn't put any Lock constraint manually, so the first element of the first constraint is locked in place. Which is the small bed mount. If you use mover to move a locked part, it will move without any constraint by default. And the whole assembly will move with it using the solver. If you use mover to move a non locked part, i.e. the big plate, then it will be constrained. PlaneAlignment will restrict your move only in that plane. Is the recommend to manually put a lock constrain on the base part

[image]

There is also an option to NOT move the locked part, by accident maybe, using this button 
[image]

akamensky, 2020-10-21:
@realthunder thanks for replying to discord question. Which I do appreciate.

This ticket is about the error I see when trying to solve constraints, not about moving the part.

realthunder, 2020-10-21:
I see. But since you post the sample file as requested, I thought this is the file that gets you the error?

akamensky, 2020-10-21:
It is, these same bodies, when I create constraints (i.e. as the one in assembly I attached) I always get error first and keep clicking on solve will `eventually` succeed.

realthunder, 2020-10-21:
The solver sometimes has trouble if the two parts are placed in awkward initial position. Use mover to nudge one of the `Non locked` part will usually help. But I just tried your parts with a fresh assembly. It seems fine. Can you please post a screencast so that I can see your steps?

[image]

akamensky, 2020-10-21:
@realthunder unfortunately animated GIF screencasts are unreliable. I've recorded a video instead. https://youtu.be/SFDa4Wc1cz4

Also IDK how you do it drag&drop so smoothly. Whenever I go to select a part to drag&drop the tree collapses (see in the video) which is pretty effing annoying... but i guess that is a separate issue.

realthunder, 2020-10-21:
Ah, I see why you have trouble now. You are constraining the bottom face of the plate, not the top. That makes the initial angle of the planes too big for the solver to solve. It is recommended to use the mover to roughly place the parts before adding the constraint.

Even if you have added constraint, you can still move it to help the solver to snap the parts. In this specific case, the reason why you can't do that is because you didn't lock the base plate explicitly (using a lock constraint on any face of the plate). So the locked part is auto determined by the first element of the first constraint, which is the small mount. And when you try to move a locked part, it will move the whole assembly using the solver. And since the solver can't solve the assembly right now, it will keep complaining.

So, the recommended workflow when creating a new assembly is to add the base part, lock it. Add subsequent parts, roughly place the parts, then start adding other constraints.

Regarding the auto collapsing tree, this is because you are in tree view auto collapse/expand mode. You can change it by right click the tree view, select `Tree view actions -> Multi document`.

akamensky, 2020-10-24:
> Regarding the auto collapsing tree, this is because you are in tree view auto collapse/expand mode. You can change it by right click the tree view, select Tree view actions -> Multi document.

Thanks for the tip, that is so much better now.

> Ah, I see why you have trouble now. You are constraining the bottom face of the plate, not the top. That makes the initial angle of the planes too big for the solver to solve. It is recommended to use the mover to roughly place the parts before adding the constraint.

I understand that solver in some conditions may have problems solving the constraints. That is just how software logic works. What I meant in this issue is the weird behavior where first attempt it fails, but second (2nd on the video, but sometimes 3rd or 4th) it succeeds. If it `eventually` succeeds, that means it can solve, which in turn means there should not be failure in the first place.

I would imagine if solver can solve it should be able to do so on the first try, if it can't solve, it should fail on every try.

ceremcem, 2021-02-22:
@akamensky Is this issue resolved?

akamensky, 2021-02-25:
Haven't tested cuz I give up on asm3. I could not use it after all.

luzpaz, 2021-06-24:
@realthunder should we close this ticket?

closed 2021-06-25 by realthunder