realthunder/FreeCAD_assembly3 #325: PartDesign Additive Loft fails with 4 or more cross-sections

BassMatiFreecad, 2020-10-19:
When I try to create a PartDesign Additive Loft feature with three sections works as expected. With four or more cross-section sketches, I get the error "NCollection?IndexedDataMap::FindFromKey". The additive loft cannot be created. No specific details in the Report View. 

See screenshot 2 and attached "additive-loft-error-2.fcstd" in the zip file.

It works fine in the Freecad mainstream version 0.19.20996, see screenshot 2 and attached "additive-loft-error-1.fcstd". When I open this file in FC Assembly 3, I get a similar error message on the Additive Loft feature.

Best regards, brilliant work so far!
Ulrich

[image]
[image]
[additive-loft-error.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5402962/additive-loft-error.zip)

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git)
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

BassMatiFreecad, 2020-10-19:
[additive-loft-error.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5402978/additive-loft-error.zip)

BassMatiFreecad, 2020-10-20:
Workaround: Split into multiple loft features with contours each. Duplicate the last sketch of the previous loft with Draft Workbench Clone2D.

realthunder, 2020-10-20:
Can you please try the latest release 2020.10.19. It seems working fine there.

BassMatiFreecad, 2020-10-20:
You did it again - it's perfect!
Big thank you & keep going!
Greetings from a German guy in Switzerland!

closed 2020-10-20 by BassMatiFreecad