realthunder/FreeCAD_assembly3 #326: Copy of Part Design Body creates a stray feature reference

BassMatiFreecad, 2020-10-20:
I'm sorry to bog again, but I'm pretty sure that this worked before:

I created a copy of a Body with a Pad feature (either via copy & paste, or with menu Edit > Duplicate selected. In the element selection dialog, I did not have to deselect anything, because the document has only this one single Body. The result is a new Body001 with a Pad001 and a Sketch001, which is fine so far. But the original Body has now two features, Pad and Pad001, where Pad001 seems to be incorrect. Both instances of Pad001 seem to reference the same object in memory, because they have the same internal name, and if I delete one, both are gone in the model tree.

[image]

Attachment: Project files (copy-of-body.zip), before & after.
[copy-of-body.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5408766/copy-of-body.zip)

Best Regards
Ulrich

luzpaz, 2021-06-29:
@BassMatiFreecad please retest on latest Daily build

Shkolik, 2021-07-13:
Tested on:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

[image]

Looks like everything works as expected.

luzpaz, 2021-07-13:
Thanks. Closing ticket

closed 2021-07-13 by luzpaz