realthunder/FreeCAD_assembly3 #327: Model Tree does not scroll when element is dragged near top or bottom border

BassMatiFreecad, 2020-10-20:


BassMatiFreecad, 2020-10-20:
Same as in mainstream version:

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22756 (Git)
Build type: Release
Branch: master
Hash: ce87f586ee36ffe8ebcdcd3947ced4ab389ae795
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2020-10-21:
I am aware of this problem in my branch. But upstream should be fine. Can you check again, in case I misunderstand your issue somehow.

BTW, there is a keyboard shortcut that is design to help dragging along large tree model. First, click to select the item(s) you want to drag. Scroll to item you want to drop. Press keyboard shortcut `T, D`, then click the target item to drop. In my current release, you can use scroll wheel while the buttonless dragging is active. In upstream, you'll need to scroll to the target item first before activate the dragging.

BassMatiFreecad, 2020-10-21:
Just re-checked: It _does_ work on the upstream. There is a significant delay (~5 seconds) on my machine (Core m3-7Y30@1.6 GHz) before it starts scrolling. This was not the case in some previous version, at least, I didn't notice.

Apologies for the false alert.
Best Regards
Ulrich

luzpaz, 2021-06-29:
@BassMatiFreecad anything left here?

closed 2021-07-30 by luzpaz