realthunder/FreeCAD_assembly3 #328: Renaming a Sketch constraint breaks formulas in model

BassMatiFreecad, 2020-10-20:
This one has been a long time around, but appears regularly:

I use the Sketch constraint "PlywoodThickness" in the Pad.Length property, like so:

[image]

The I change the constraint name in the Sketch to "OLDPlywoodThickness":

[image]

Then I click the "Update" button (just to be sure...) and close the sketch. When I check the Pad.Length property, the constraint name has been changed correctly, but _the Sketch name was lost_:

[image]

Recompute and/or reloading the document does not find and flag the broken formula(s), no errors in the report view - which may cause some serious headache later. The broken formula is not detected until the formula editor is opened:

[image]

Using the label name <<S000>> instead of the internal name "Sketch" has the same effect.

Might be somewhat related to the old [#311](https://github.com/realthunder/FreeCAD_assembly3/issues/311).

Attachment: 
[workdesk-1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5410222/workdesk-1.zip)

Best regards
Ulrich

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.10.19.22739 +2669 (Git)
Build type: Release
Branch: LinkStage3
Hash: 413465d933e94061fb29486f3a322220e2604492
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

BassMatiFreecad, 2020-10-20:
Mainstream version of FreeCAD has the same problem. Should I rather post such problems in the github/Freecad project? Then I won't bother you with problems that is caused outside of ASM3...

Regards
Ulrich

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22756 (Git)
Build type: Release
Branch: master
Hash: ce87f586ee36ffe8ebcdcd3947ced4ab389ae795
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

ceremcem, 2020-10-20:
> Mainstream version of FreeCAD has the same problem. Should I rather post such problems in the github/Freecad project? Then I won't bother you with problems that is caused outside of ASM3...

We have an [`upstream` label](https://github.com/realthunder/FreeCAD_assembly3/labels/upstream) and realthunder puts that label when necessary. As an ASM3 user, I'm happy to be informed about such critical bugs by *this* channel.

BassMatiFreecad, 2020-10-20:
Excellent! Then I'll keep posting here, because I also work exclusively with ASM3, because I believe that ASM3 is somewhat ahead of the upstream FreeCAD. Only when I find bugs I verify that with the latest upstream version, if possible.

realthunder, 2020-10-21:
Should be fixed. I'll make a release today.

closed 2020-10-21 by realthunder