realthunder/FreeCAD_assembly3 #329: Creating a mesh in the 'Mesh Design' workbench is broken and gives error in 22/10 appImage

dreetie, 2020-10-22:
**How to recreate issue:** make any shape (cube in part workbench)
go to 'Mesh design' workbench and tesselate shape. export fails.

**report view:**

`10:23:39  /tmp/.mount_FreeCAW0634U/usr/lib/python3.8/site-packages/mpmath/ctx_mp_python.py:892: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if other is 0:
10:23:39  /tmp/.mount_FreeCAW0634U/usr/lib/python3.8/site-packages/mpmath/ctx_mp_python.py:986: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if other is 0:
10:24:30  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ImportError'>: /tmp/.mount_FreeCAW0634U/usr/lib64/../lib/libNETGENPlugin.so.8.3: undefined symbol: _ZN6netgen11NgExceptionD1Ev
10:24:46  Traceback (most recent call last):
  File "<string>", line 5, in <module>
<class 'NameError'>: name 'MeshPart' is not defined
10:24:46  name 'MeshPart' is not defined`

**AppImage version:** FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201022-glibc2.12-x86_64.AppImage
[image]

realthunder, 2020-10-22:
@looooo Hi, I am always having trouble with the `netgen` and `smesh` in conda. Conda often seems to pick the wrong version. I tried to pin to a specific version before. But then it leads me to have to pin more and more packages. Any tricks?

looooo, 2020-10-22:
@realthunder 
The latest combination doesn't work properly. best to use this version: https://github.com/FreeCAD/FreeCAD-AppImage/blob/master/conda/linux/create_bundle.sh#L4

I will work on updating the dependency in wintertime. This is mainly boost74, python3.9, vtk9, occt3.5. Once we are back on top of dependencies things should be less restricting.

dreetie, 2020-10-29:
@realthunder New build works. No I don't use OBJ format. Only stl because I need to export for 3D printing.

psi-4ward, 2023-04-14:
ArchLinux here.

So, I run into `Unknown error occurred while running OpenSCAD`. Problem is trivial to solve: Just load the **OpenSCAD** Workbench and Mesh is working again. But `import MeshPart` didnt not work caus of different missing Libs. Tried to install them but ended up using the AppImage