realthunder/FreeCAD_assembly3 #330: BUG: Unable to create constraints

akamensky, 2020-10-25:
I would consider this as a bug since the expected workflow (based on tool description) did not work for me at all.

# Setup
```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22756 (Git)
Build type: Release
Branch: master
Hash: ce87f586ee36ffe8ebcdcd3947ced4ab389ae795
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)
```

# Overview
9 bodies assembly:
- 1 base body that is locked (body num. 1)
- 4 bodies are "plane aligned" to base body at an offset (thus leaving some dof's) (num. 2-5)
- 4 newly added bodies (no constraints yet) (num. 5-9)

Multiple scenarios failed so far:

## Scenario 1
Add one more constraint between base body and one of 4 partially constrained bodies -- axial alignment between a round hole of base body and one of cylindrical surfaces in body num.2. Cylindrical surface is at 90deg to "aligned surface", thus already making axels parallel, but attempting to solve constraint results in "inconsistent constraints" error.

## Scenario 2
Add a constraint between body num.2 and body num.5 for face alignment. Depending on specifics the constrain either produces "inconsistent constraints" or "not converging". Both of which are so non-descriptive that it is impossible to deduce what steps could one take to fix this.

Both scenarios are in assembly attached below (constraints are created, but disabled). Assembly file is "3. assemblies/1. Bed assembly.FCStd"

[HEVO2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5434606/HEVO2.zip)

realthunder, 2020-10-25:
PlaneAlignment (-2 rotational, -1 positional) + AxialAlignment ( -2 rotational, -2 positional) on the same pair of parts causes over constraint.

You can use a single PlaneCoincidence to achieve the same effect. If you don't want the mount to be rotatable around the hole, then you can either set PlaneCoincidence.LockAngle or change to Attachment constraint (prefered). The screencast below shows how to flip the part. The same menu also has action to offset the element/part.

[image]

luzpaz, 2021-07-13:
No response from @akamensky. Closing ticket

closed 2021-07-13 by luzpaz