realthunder/FreeCAD_assembly3 #332: Errors when recomputing constraints

brychanrobot, 2020-10-31:
My assembly must have something broken, but I can't figure out what it is. A week ago I was able to add constraints and have things recompute and adjust, but today I started getting python errors when recomputing constraints, and even if the constraint does seem to work, the elements are always highlighted in red. There are different errors for different constraints, but the one I get for PlaneCoincident is 
```_lambdifygenerated() missing 6 required positional arguments: 'Dummy_129', 'Dummy_130', 'Dummy_132', 'Dummy_128', 'Dummy_127', and 'Dummy_126'```

[image]

I've tried this with both the 0818 release and the 1026 release. They both exhibit the same behavior, so I'm convinced that there's something funny going on with my assembly. Another thing to note is that I was able to execute this exact same constraint on these same parts a week ago without the odd behavior.

I've attached the assembly in case that helps.
[example_assembly_with_errors.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5469502/example_with_errors.zip)

brychanrobot, 2020-10-31:
The full traceback is:

```
16:53:57  34.901585 <asm3.main> assembly.py(3644): solver exception when auto recompute
Traceback (most recent call last):
  File "<string>", line 519, in _catch
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/solver.py", line 416, in solve
    return _solve(*args,**kargs)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/solver.py", line 386, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/solver.py", line 109, in __init__
    self.system.solve(group=self.group,reportFailed=reportFailed)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/sys_sympy.py", line 1283, in solve
    ret = sopt.minimize(self.F,x0,(eq,jeqs,heqs), jac=jac,hess=hessF,
  File "/tmp/.mount_FreeCA9Mzxdy/usr/lib/python3.8/site-packages/scipy/optimize/_minimize.py", line 600, in minimize
    return _minimize_powell(fun, x0, args, callback, **options)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/lib/python3.8/site-packages/scipy/optimize/optimize.py", line 2631, in _minimize_powell
    fval = squeeze(func(x))
  File "/tmp/.mount_FreeCA9Mzxdy/usr/lib/python3.8/site-packages/scipy/optimize/optimize.py", line 327, in function_wrapper
    return function(*(wrapper_args + args))
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/sys_sympy.py", line 1086, in F
    res = eq(*params)
TypeError: _lambdifygenerated() missing 6 required positional arguments: 'Dummy_275', 'Dummy_274', 'Dummy_272', 'Dummy_271', 'Dummy_270', and 'Dummy_269'
```

brychanrobot, 2020-10-31:
And if I try to constrain the same faces with PlaneAlignment, I get:
```
16:53:22  44.671219 <asm3.main> assembly.py(3644): solver exception when auto recompute
Traceback (most recent call last):
  File "<string>", line 519, in _catch
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/solver.py", line 416, in solve
    return _solve(*args,**kargs)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/solver.py", line 386, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/solver.py", line 109, in __init__
    self.system.solve(group=self.group,reportFailed=reportFailed)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/sys_sympy.py", line 1171, in solve
    eq = o.getEqWithParams(params)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/sys_sympy.py", line 411, in getEqWithParams
    return self.getEq()
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/sys_sympy.py", line 807, in getEq
    return _pointPlaneDistance(self.pt,self.pln)
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/sys_sympy.py", line 447, in _pointPlaneDistance
    e = _project(pln,[pt])
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/sys_sympy.py", line 440, in _project
    return [ e.Vector.dot(r.i)+e.Vector.dot(r.j) for e in args ]
  File "/tmp/.mount_FreeCA9Mzxdy/usr/Mod/asm3/freecad/asm3/sys_sympy.py", line 440, in <listcomp>
    return [ e.Vector.dot(r.i)+e.Vector.dot(r.j) for e in args ]
AttributeError: 'list' object has no attribute 'Vector'
```

realthunder, 2020-11-01:
You have accidentally changed the solver type of your assembly. Just set it back to SolveSpace by change the `SolverType` property of the assembly. The sympy solver is outdated. I should probably disable it until I have time to make it work again.

luzpaz, 2022-03-21:
@realthunder closing this issue