realthunder/FreeCAD_assembly3 #336: OpenSCAD Qt library linkage, OpenSCADUtils package import.

ian-arkver, 2020-11-10:
First off - thanks @realthunder for the continuing hard work on this branch.

I had failures importing an OpenSCAD file with the nightly build from 20201106, running on up-to-date Manjaro Linux. FreeCAD calls out to the system OpenSCAD installation in /usr/bin which is linked against Qt 5.15.1 but the AppImage is built with Qt 5.12.9. OpenSCAD fails to start due to missing libraries since LD_LIBRARY_PATH allows only AppImage paths.

I worked around this by unpacking the AppImage and editing AppRun to put system paths first. A bit hacky, but it allowed OpenSCAD to start up.

I then hit another problem with the import module path for OpenSCADUtils. This is imported in a few places in usr/Mod/OpenSCAD/OpenSCADFeatures.py and is normally a plain import OpenSCADUtils with imported classes and members prefixed only with "OpenSCADUtils."

However at line 502 there's an import freecad.OpenSCAD.OpenSCADUtils and a call to freecad.OpenSCAD.OpenSCADUtils.process_ObjectsViaOpenSCADShape. This path is not found and import fails.

I changed the import and call to match the others in this file and then it works. I was able to import my OpenSCAD model file correctly.

Hope this is helpful. All the best,
Ian

realthunder, 2020-11-12:
Hi, can you please post this to freecad forum, maybe [this](https://forum.freecadweb.org/viewtopic.php?f=42&t=40085) thread? I see the line 502 change is very recent. It used to be without the `freecad` prefix. They must have done it for a reason I guess.

ian-arkver, 2020-11-12:
Thanks for the pointer to the thread. Looking at the github source in the OP there, I think my second point has already been fixed (changed anyway - will try it):
https://github.com/FreeCAD/FreeCAD/commit/f3bdaaa55a6c03b297924c40819d23e4603fa55b

I think I can get around the Qt library problem by pointing Freecad at a wrapper script for OpenSCAD which overrides the LD_LIBRARY_PATH just for it, so don't need to mess with the AppRun within the appimage.

Anyhow, I'll close this issue and if it still doesn't work I'll post over there.

All the best,
Ian

closed 2020-11-12 by ian-arkver