realthunder/FreeCAD_assembly3 #338: How do we assemble an object with its mirror?

ceremcem, 2020-11-12:
# Description 

Here is an example shape I want to assemble with each other: 

[image]

When I click an entity on the mirrored object, constraints on the menu are available. When I click an entity on the original object, constraints are grayed out, so I can not select two entities: 
 
[image]

What is wrong with this workflow?

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24494 (Git)
Build type: Release
Branch: LinkDev
Hash: 5c756018210aede00673c0559933f072bbbadb40
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

[mirror-problem.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5532093/mirror-problem.FCStd.zip)

> Note: Remove the `.zip` extension

realthunder, 2020-11-12:
It's usually not a good idea to mix Part and PartDesign feature, especially when coupled with asm3 here. The reason of the gray out is because the child feature of the mirror occupies the exact same place as the the non-mirrored body in the assembly. In your case, this child feature is the same feature as the non-mirrored body. But to asm3, they are treat as different object. The child feature is not considered inside the assembly container, so as the gray out. Pay attention to the tree view item selection reaction.

[image]

Part Mirror (or any Part feature for that matter), will take account the child feature placement, and adds its own placement on top of it. So even if you move out the assembly as shown in the above screen cast, your assembly will not work. Because move the non-mirrored body will indirectly move the mirror as well, which will confuses solver for sure.

You have a few options. 

1) Change the mirror to a link with negative scale
2) Or, right click the child feature of the mirror, and choose `Link actions -> Replace with link`, which replaces the child feature with a link
3) Or, right click the non-mirror body in the assembly, and do `Replace with link`. You can hide the child feature to avoid confusion.

In case you want to do further model with the mirror, or the original body, you can also

4) Create a Link to the non-mirror body, and use the Link as base feature to create another body.
5) Or, if you are using my latest revision, you can create the mirror in the same body as `New solid`, which gives you separate solid for the non-mirrored and mirrored feature. Create a Link to both the non-mirrored and mirrored feature inside the body, and add them to assembly instead.

bambukouk, 2021-04-30:
I don't know if this is the right place to comment/ask questions, or should I start separate issue?
or perhaps raise on FreeCAD forum?
If I got it wrong - sorry...

Anyway here it goes:
I have tried following your instructions at:
5.Or, if you are using my latest revision, you can...etc

It all seem fine (see attached file for sanity check, please), but I end up with 3 bodies for each mirrored pair?
It's fine for a simple model, but for a complex one with many mirrored bodies it is a bit of a mess :-(
or, have I got it wrong?

Which leads me to another question, please:
Can one use mirrored sub-assemblies in assembly?

Thank you, again
[link(mirror).zip](https://github.com/realthunder/FreeCAD_assembly3/files/6405036/link.mirror.zip)

luzpaz, 2021-07-13:
@ceremcem are you awaiting your own response here? Or am i missing something :confused: :sweat_smile:

bambukouk, 2021-07-13:
Strangely, I got copied on your comment @luzpaz as a "commenter" for this issue but i never took part in this discussion?
Having said so, as far as I am concerned @realthunder reply has answered the question (at least for me) and the issue could be closed?

luzpaz, 2021-07-13:
@bambukouk are you watching this whole repo? if so, then that would explain it

bambukouk, 2021-07-13:
Yes, I am watching it closely 😀
but I can't be the only one?
Anyway - it's no issue, just curiosity on my part...

ceremcem, 2021-07-16:
> @ceremcem are you awaiting your own response here? Or am i missing something confused sweat_smile

Yeah, it's kind of... Yeah, I think so :) I re-read the conversation and probably I had confused which path to follow and I didn't want to mark as "resolved" or something else without trying it out. Anyway, I got the message :)

luzpaz, 2021-07-30:
Closing. If further discussion merited, please re-open

closed 2021-07-30 by luzpaz