realthunder/FreeCAD_assembly3 #339: Coordinate system: planes and axis

parnasist3, 2020-11-13:
When I finally could create a (datum)point and reference it to "Objects origin" I there after created an axis/coordinate system connected to that (datum)point.
I then thought this will be a walk in the park so I just wanted to create a plane offset from one of the axis/coordinate systems planes, but to my big surprise there were no planes to offset from! And no axis either. Where did they go?

[image]

realthunder, 2020-11-15:
The 'coordinate system' you have created is not the same object as 'Origin' used in Part/Body container. The 'Origin' object itself has no placement, and therefore cannot be moved. Only the container that owns the origin can. 

The 'coordinate system' is in fact a type of datum object. Similar to datum point/line/plane. If you want to attach to different axis plane, simply select the corresponding normal axis when attaching. For example, select the red axis (i.e. the X axis) allows you to attach to YZ plane.

parnasist3, 2020-11-15:
Since my ultimate goal were to get a static non-movable / -erasable / -rotationable but hidable coordinate system at the top / root of the file, I then will request a coordinate system with the same properties as "The 'Origin' object itself has no placement, and therefore cannot be moved."
Further I request that every new file the user create shall have this coordinate system and point at creation time. I.e. it shall not be possible to opt it out. It shall how ever be possible to set a default hide/show value in the settings, for those who whish that. It shall of course also be simple to hide/show it from the normal gui.
May I suggest you make a combined point and axis and plane symbol? And make it work as such as well.

realthunder, 2020-11-17:
Non-movable but rotatable? Can you elaborate more on this? The `Origin` object cannot be rotated.

> May I suggest you make a combined point and axis and plane symbol? And make it work as such as well.

There is indeed an object like this, it is called `App::Placement`. Not sure if this is what you want. You can add it like this 

```
App.ActiveDocument.addObject('App::Placement', 'placement')
```

The attach engine does not support this object yet. But I can add support for that. How do you plan to use this, through attachment?

parnasist3, 2020-11-22:
The point _and_ coordinate system _and_ planes shall not be possible to: move, rotate, delete, replace. But they can be hidden and be colored differently.
Everything created in the part file shall **somehow** relate back to this point _and_ coordinate system _and_ planes.
The point _and_ coordinate system _and_ planes will be the thing one can trust 100%, what ever happens, in the part file.

parnasist3, 2020-12-09:
**asm3 20201208:**

Regarding sketch versus offset planes, it seams the sketch does not obey the mapping I set, see picture.
[image]

[Sketch_attach_vs_offset_mapped_plane.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5665084/Sketch_attach_vs_offset_mapped_plane.FCStd.zip)
 (remove .zip at the file name end)

realthunder, 2020-12-12:
The map mode you chose is `TagentPlane`, which is probably not what you want. And according to its definition, the attacher probably should just throw an error or something, because there is no tangent plane that satisfy your selection.

BTW, what you are doing seems like exactly what App::Part provides. Why not just use App::Part instead?

[image]

parnasist3, 2020-12-15:
asm3 20201214:

Regarding plane offset: so there is no command "offset plane with respect to user chosen plane"? And keep it connected to chosen plane.?

Also noticed that it is possible to make sketch stay on the user chosen plane if I first select the plane then the point, really odd behavior. Do not get the logic of a sketch being attaching to a point and not on the plane just because of selection order. The point should of course, auto-magically, be projected on to the plane then attaching the sketch to both the plane and the point.

We seam to talk/type past each other, so I created this example:
[image]

[Sketch_attach_vs_offset_mapped_plane_3.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5695364/Sketch_attach_vs_offset_mapped_plane_3.FCStd.zip)

Here you see how one build up a model (could not fulfill the mission due to missing functionality).
One starts with a new file, which should already contain a file-zero, i.e. point, coordinate system and planes.

 **1a)** Then you create a new point (WorkinCoordinateSystem, this will double as Turbo inlet reference) in space with respect to file zero, close to where you will create your thingie, in this case a turbo tube.
 **1b)** Create a new coordinate system attached to the newly created point (WorkinCoordinateSystem). This coordinate system should contain the planes already.
 **1c)** Create a new point (TtHO-point) for Turbo_tube_Hose_connection_Origin.
 **1d)** Create a line (Z-line_1) with its start point attached to Turbo_tube_Hose_connection_Origin-point and direct it upward and parallel to WorkinCoordinateSystem Z-axis _(could not fulfill that with this edition of asm3)_.
 **1e)** Create a plane (Z-rotation_plane) through the newly created line and set it at an angle with respect to WorkinCoordinateSystem's ZX-plane _(could not fulfill that with this edition of asm3, I therefor created a coordinate system just for the sake of visualization/readability)_.
 **1f)** Attach a sketch on Z-rotation_plane and TtHO-point and orient its vertical axis along Z-line_1 _(could not fulfill that with this edition of asm3, I therefor created a coordinate system just for the sake of visualization/readability)_.
 **1g)** In the sketch create a line (Z-line_2) angled with respect to the sketch vertical axis. This will be your new vertical axis to be used from outside the sketch.
 **1h)** Create a plane through Z-line_1 and Z-line_2). Now we have the basis for a Turbo tube. This model can now be reused in a very simple way: just make a copy of the file and move the point in WorkinCoordinateSystem and EVERYTHING moves with it ! Then move TtHO-point and reorient Z-line_2, et voilá, done n'dusted! A completely new model with almost no effort.

 **2a)** The "Master_Body" should already be in the file when created. But since it wasn't I created a new body.
 **2b)** Create yet another body.
 **2c)** Create a sketch in the newly created body attaching it to WorkinCoordinateSystem and its point plus orient sketch horizontal axis parallel to WorkinCoordinateSystem Y-axis _(could not fulfill the orientation bit with this edition of asm3)_.
 **2d)** Create a profile to rotate and make a solid.
 **2e)** Add the body to "Master_Body" _(could not fulfill the add bit with this edition of asm3)_. As I see it, the body container lacks this feature/function: act as a container for other bodys. Since this is essential for the multi-body approach to work, I would request it. Perhaps it already exist? Maybe I was searching for elephants?

realthunder, 2020-12-17:
> Regarding plane offset: so there is no command "offset plane with respect to user chosen plane"? And keeping it connected to chosen plane.?

There is. After activating the attacher editor task, just scroll down to reveal the offset settings.

[image]

> **1h)** Create a plane through Z-line_1 and Z-line_2). Now we have the basis for a Turbo tube. This model can now be reused in a very simple way: just make a copy of the file and move the point in WorkinCoordinateSystem and EVERYTHING moves with it ! Then move point_2 and reorient Z-line_2, et voilá, done n'dusted! A completely new model with almost no effort.

The workflow you are using is quite _anti-FreeCAD_, so to speak. Make a copy of the entire file and adjusting the point just to get the same model at a different location is really not that convenient. App::Part is designed for this purpose. Please check out [this video](https://youtu.be/5vzgrP95ris) to get some ideas.

Some key points

1) when you add a body to an App::Part, the body is implicitly attached to App::Part's origin, so when you move the App::Part, the body inside moves with it. There are better way to reuse the model when using App::Part. You can create App::Link to link the part from other files without copying anything.

2) Body itself can be attached to other geometry. This is equivalent to attach the body's origin to another geometry. Same thing for the App::Part, it can also be attached to other geometry at higher hierarchy. Right now, the `Attachment...` action is only available in Part workbench, but it does work on various objects in other workbench. Make sure you select the body, before activate the attachment.

3) You must not attach anything inside a body to something outside of the body. In most cases, it will be enough to attach body itself to other geometry, and attach feature inside the body to other features in the same body. If you really want to attach something outside, use SubShapeBinder (the green one) to _import_ the shape into the body, and then attach to the binder instead.