realthunder/FreeCAD_assembly3 #340: Feature request: Show dialog box when using additive or subtractive tools with geometries which are out of scope

Rabbit-sk, 2020-11-13:
Dear Realthunder,

Although it is possible in Part Design workbench to include in active body geometries which are out of it, I usually don't want to do that. To avoid doing so by mystake, I would appreciate dialog box showing up after icon of some additive or subtractive tool is clicked, and if selected geometry is out of scope. The statement could be: "Selection is not in active body. Do you want to continue?" The highlighted option should be "No". Something simmilar to this dialog box:

[image]

Regards,
Ján

PS: Thank you for quick fix of #337 bug.

Rabbit-sk, 2020-11-13:
[image]

realthunder, 2020-11-16:
This is considered a bug. I have fixed it by brought back the old dialog asking user to choose how to import the sketch to the active body. However, this type of workflow is deprecated in my branch. I haven't got time to find a better way to handle it. The recommended way is to use the green `SubShapeBinder` to import any type of object into the active body, and then create feature based on the binder.

closed 2020-11-16 by realthunder

Rabbit-sk, 2020-11-16:
I allways use green SubShapeBinder, but when i found out that it is not necessary, I was wondering if that is bug or feature. Thank you for clarification. 

When user is going to do something in a wrong way, FreeCAD usually shows error, warning or options how to continue. For more convenient work I think it would be sufficient to display some warning message box, because warnings in report view could be easily unnoticed.

It is possible to use warning like this (which contains string "out of scope") from report view:  

<App> GeoFeatureGroupExtension.cpp(571): Nepomenovan__#Pad.Profile(links to Nepomenovan__#Sketch) is out of scope:Nepomenovan__#Body vs. Nepomenovan__#Body001

to just show some warning message box with message "Out of scope" and OK button, after icon of some additive or subtractive tool is clicked?