realthunder/FreeCAD_assembly3 #341: Constraint multiplication error

lilaL, 2020-11-14:
I'm trying to multiply a constrain and get the following message:
```
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAWMgmw7/usr/Mod/asm3/freecad/asm3/assembly.py", line 2433, in execute
    self.checkMultiply()
  File "/tmp/.mount_FreeCAWMgmw7/usr/Mod/asm3/freecad/asm3/assembly.py", line 2321, in checkMultiply
    pos0 = info0.Placement.multVec(
<class 'AttributeError'>: 'NoneType' object has no attribute 'Placement'
```
[image]

You find the project file [here](https://office.deltaempower.org/s/qQE5b6PYakMk5zk). It's in the `SubAssembly.WindVane`, you see the important elements in the following screenshot:
[image]

I wonder if it's also related to #197 ?

OS: Manjaro Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.7.8
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0

realthunder, 2020-11-16:
I have just made a new release containing some fix of the problem in this and the other issue you reported. Not sure if it is a complete fix, so I didn't close the issues yet. Multiplication requires the base part (i.e. the non-multiplied part) to be unfreezed. The multiplied part can be frozen. My fix now have proper error message if you try to do it with frozen base part. 

BTW, one of your holes in the base has a unique radius. Multiplication can only auto expand to circular edges of equal radius. You can of course drag in more circular edges to the constraint for manual expansion.

[image]

lilaL, 2020-11-17:
Thanks for fixing it!
And yeah I forgot to set the proper radius!

closed 2021-12-30 by realthunder