realthunder/FreeCAD_assembly3 #345: loss of thread size after closing and reopening the file

Grubuntu, 2020-11-19:
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.11.16.22739 +2766 (Git)
Build type: Release
Branch: LinkStage3
Hash: 6730ad3b3bd841c2745ec98bf953c523dea86913
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

Hello 

When I create threaded holes in Part Design (ISOmetric thread) the value of thread size still return at M1.6 after saving , closing and reopening the file.

Before closing the file 👍 
[image]

After reopening the file
[image]

[test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5566094/test.zip)

realthunder, 2020-11-26:
I've done a merge with upstream. The problem seems fixed already. Please check.

Grubuntu, 2020-11-26:
Thank you realthunder, its solved.

closed 2020-11-26 by Grubuntu