realthunder/FreeCAD_assembly3 #346: Spreadsheet demo does not work anymore

lilaL, 2020-11-20:
The [BOM demo](https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet#demonstration) is not working for me so I refined yours:
```
##@@ A1 bom#Spreadsheet.cells (Spreadsheet)
##@@
<<Please select document:>>

##@@ A2 bom#Spreadsheet.cells (Spreadsheet)
##@@
<<Please select assembly:>>

##@@ A3 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="Populate" editMode="1" editModeName="Button" />
def Populate():
    #@pybegin
    rstart = 5
    r = rstart
    count = 0
    try:
        while ._self.get('A' + str(r)) :
            count += 1
            r += 1
    except:
        pass

    if count: ._self.removeRows(str(rstart), count)

    r = rstart
    asm = .B2[0][.B2[1]]
    parts = asm.Proxy.getPartGroup()
    for obj in parts.Group :
        for _, col, func in .B3[1] :
            addr = col + str(r)
            ._self.set(addr, "'" + func(obj))

        r += 1

##@@ A4 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell editMode="3" editModeName="Label" />
[<<#>>, ._self.column(), lambda obj : .countLinks(obj)]

##@@ B1 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell editMode="2" editModeName="Combo" />
[._app.listDocuments(), <<None>>]

##@@ B2 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell editMode="2" editModeName="Combo" />
[{ obj.Label : obj for obj in ._app.getDocument(.B1[1]).Objects if .isAssembly(obj) }, <<None>>]

##@@ B3 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell editMode="3" editModeName="Label" />
[<<template>>, [*A4:E4]]

##@@ B4 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell editMode="3" editModeName="Label" />
[<<Document>>, ._self.column(), lambda obj : obj.Document.Name]

##@@ C2 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="isAssembly" />
def isAssembly(obj):
    #@pybegin
    return str(getattr(obj, 'Proxy', None)).startswith('<freecad.asm3.assembly.Assembly ')

##@@ C4 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell editMode="3" editModeName="Label" />
[<<Label>>, ._self.column(), lambda obj : obj.Label]

##@@ D2 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="partInfo" />
def partInfo(obj, key):
    #@pybegin
    try:
        return eval(obj.Label2).get(key, <<?>>)
    except:
        return <<?>>

##@@ D4 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell editMode="3" editModeName="Label" />
[<<Part No.>>, ._self.column(), lambda obj : .partInfo(obj, <<pn>>)]

##@@ E2 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="countLinks" />
def countLinks(obj):
    #@pybegin
    try:
        return str(obj.ElementCount + 1)
    except:
        return <<?>>

##@@ E4 bom#Spreadsheet.cells (Spreadsheet)
##@@<Cell editMode="3" editModeName="Label" />
[<<Description>>, ._self.column(), lambda obj : .partInfo(obj, <<desc>>)]
```

realthunder, 2020-11-20:
Yeah, I know. I am actually in the middle of updating the wiki. You can try the new code after I made the next release.

realthunder, 2020-11-20:
Oh, didn't realize you've made you own change. Good for you!

lilaL, 2020-11-20:
I wonder if your're planning a BOM listing tool?

snelweg, 2020-11-24:
Would be really great to have a BOM!

luzpaz, 2021-07-17:
Closing this ticket
cc @realthunder

closed 2021-07-17 by luzpaz