realthunder/FreeCAD_assembly3 #347: Crash while using FCCamera macro

ceremcem, 2020-11-25:
I wanted to try out the [FCCamera](https://wiki.freecadweb.org/index.php?title=Macro_FCCamera) macro. In the first attempt, it required the missing `PySide2` dependency. I installed it:

```
$ sudo pip install pyside2
[sudo] password for aea: 
Collecting pyside2
  Downloading https://files.pythonhosted.org/packages/5d/3b/49dbaf8910a4983c7ba08c342fdadada5f909206ce0dce88ec4aa203e7f3/PySide2-5.15.2-5.15.2-cp27-cp27mu-manylinux1_x86_64.whl (163.7MB)
    100% |████████████████████████████████| 163.7MB 7.1kB/s 
Collecting shiboken2==5.15.2 (from pyside2)
  Downloading https://files.pythonhosted.org/packages/e5/9c/27087f157e87b0cad40cce83c63ed157742def099d16882468569b01f06d/shiboken2-5.15.2-5.15.2-cp27-cp27mu-manylinux1_x86_64.whl (1.2MB)
    100% |████████████████████████████████| 1.2MB 248kB/s 
Installing collected packages: shiboken2, pyside2
Successfully installed pyside2-5.15.2 shiboken2-5.15.2
```

# Traceback
This is the traceback when I hit the "Run macro" button: 
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7f763a2c8060]
#1  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyType_Ready+0x11) [0x7f7641dcd331]
#2  /usr/local/lib/python2.7/dist-packages/shiboken2/libshiboken2-python2.7.so.5.15(+0x2d0dc) [0x7f75ac3680dc]
#3  /usr/local/lib/python2.7/dist-packages/shiboken2/libshiboken2-python2.7.so.5.15(+0x2cf40) [0x7f75ac367f40]
#4  /usr/local/lib/python2.7/dist-packages/shiboken2/libshiboken2-python2.7.so.5.15(FinishSignatureInitialization+0x17) [0x7f75ac366ac7]
#5  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(_PyImport_LoadDynamicModule+0x9c) [0x7f7641e71f7c]
#6  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x189da1) [0x7f7641e72da1]
#7  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyImport_ImportModuleLevel+0x1ee) [0x7f7641e7410e]
#8  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x89328) [0x7f7641d72328]
#9  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7641dbf5c3]
#10  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47) [0x7f7641ed26c7]
#11  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x277a) [0x7f7641d662ca]
#12  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7641ed329c]
#13  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19) [0x7f7641d639c9]
#14  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyImport_ExecCodeModuleEx+0x94) [0x7f7641e70c64]
#15  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x188650) [0x7f7641e71650]
#16  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x1895c1) [0x7f7641e725c1]
#17  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x189da1) [0x7f7641e72da1]
#18  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyImport_ImportModuleLevel+0x515) [0x7f7641e74435]
#19  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x89328) [0x7f7641d72328]
#20  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7641dbf5c3]
#21  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47) [0x7f7641ed26c7]
#22  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x277a) [0x7f7641d662ca]
#23  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7641ed329c]
#24  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d) [0x7f7641d6a25d]
#25  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7641ed329c]
#26  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19) [0x7f7641d639c9]
#27  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyImport_ExecCodeModuleEx+0x94) [0x7f7641e70c64]
#28  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x188650) [0x7f7641e71650]
#29  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x1895c1) [0x7f7641e725c1]
#30  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x189da1) [0x7f7641e72da1]
#31  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyImport_ImportModuleLevel+0x1ee) [0x7f7641e7410e]
#32  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x89328) [0x7f7641d72328]
#33  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7641dbf5c3]
#34  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47) [0x7f7641ed26c7]
#35  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x277a) [0x7f7641d662ca]
#36  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7641ed329c]
#37  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19) [0x7f7641d639c9]
#38  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyRun_FileExFlags+0x8e) [0x7f7641e1b03e]
#39  0x7f7642b09195 in Base::InterpreterSingleton::runFile(char const*, bool) from /opt/FreeCAD/lib/libFreeCADBase.so+0x95
#40  0x7f7643b31902 in Gui::MacroManager::run(Gui::MacroManager::MacroType, char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x1d2
#41  0x7f7643bca4ca in Gui::Dialog::DlgMacroExecuteImp::accept() from /opt/FreeCAD/lib/libFreeCADGui.so+0x1da
#42  0x7f763b6cf660 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x4d0
#43  0x7f763c820c72 in QAbstractButton::clicked(bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x42
#44  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x59a5c3) [0x7f763c5575c3]
#45  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x59b944) [0x7f763c558944]
#46  0x7f763c558a54 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x74
#47  0x7f763c1d7e80 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x290
#48  0x7f763c18154c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#49  0x7f763c189ca7 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a47
#50  0x7f7643b2f0a1 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
[... 33 lines cut ...]
#84  0x7f763b6eb854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#85  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7f763c22b5d6]
#86  0x7f763b6b97ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#87  0x7f763b6b9b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#88  0x7f763b6bfbd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#89  0x7f7643a89c06 in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x1d56
#90  freecad-git(+0x367a) [0x5560cd0e167a]
#91  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f763a2b52e1]
#92  freecad-git(+0x47ca) [0x5560cd0e27ca]
```
# Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24494 (Git)
Build type: Release
Branch: LinkDev
Hash: 5c756018210aede00673c0559933f072bbbadb40
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2020-11-25:
You'll need Python3 and qt5 for that. Also please sync to branch LinkStage3.

ceremcem, 2020-11-25:
I synced LinkStage3 branch to: 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25920 (Git)
Build type: Release
Branch: LinkDev
Hash: fd4fa30ab07d30e3bcea353086278bcb1b05c83d
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

However, I don't know how to point Python3 for FreeCAD because Python3 is already the default Python version in my system: 

```
# update-alternatives --config python
There are 2 choices for the alternative python (providing /usr/bin/python).

  Selection    Path                Priority   Status
------------------------------------------------------------
* 0            /usr/bin/python3.5   2         auto mode
  1            /usr/bin/python2.7   1         manual mode
  2            /usr/bin/python3.5   2         manual mode
```

I tried to remove build directory and build from scratch, but here is the log at the beginning of the build process: 

```
-- Using default python: -python2.7
```

I don't get why FreeCAD thinks that Python2 is the standard. 

# Current status 
Syncing branch didn't helped and FreeCAD crashed with exactly same traceback.

realthunder, 2020-11-26:
You'll need to rerun CMake, with some additional argument, such as `-D PYTHON_EXECUTABLE:FILEPATH=$PREFIX/bin/python` `-D BUILD_QT5:BOOL=ON`, where `$PREFIX` points to your python3 installation location. Check [here](https://github.com/realthunder/FreeCADMakeImage/blob/master/conda/freecad_asm3/build.sh) for reference.

BTW, maybe you want to try my build script [here](https://github.com/realthunder/FreeCADMakeImage). To automate a build of AppImage with conda, use the following command. You'll need to install `docker` for it to work.

```
./mkasm.sh conda sudo
```

ceremcem, 2020-12-01:
The problem has been gone with the current build.

closed 2020-12-01 by ceremcem