realthunder/FreeCAD_assembly3 #350: Exception while switching to Draft WB

ceremcem, 2020-11-26:
The following exception occurs when I switch to Draft WB:

[image]

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25920 (Git)
Build type: Release
Branch: LinkDev
Hash: fd4fa30ab07d30e3bcea353086278bcb1b05c83d
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2020-11-27:
Looks like an incompatible pivy (the Coin Python binding) version used. Did you make any change to pivy or Coin?

ceremcem, 2020-11-27:
No, there were absolutely no changes with my setup (it's the same LXC container I've been building and running FC). 

I've finally managed to start the compilation (natively on Debian Buster) by using your build scripts. I hope I can give a new feedback with the newly generated AppImage in a few ours.

realthunder, 2020-11-28:
I also use LXC for Python 3 build. Why not just building using a ubuntu 18.04 LXC image. You can follow the standard FreeCAD build [instruction](https://wiki.freecadweb.org/Compile_on_Linux#Python_3_and_Qt5) for that. It'll be faster than building the AppImage. My cmake argument for that is simply. You may want to change CMAKE_BUILD_TYPE to `Release`.

```
cmake -DCMAKE_BUILD_TYPE=Debug -DFREECAD_USE_OCC_VARIANT=Official -DBUILD_QT5=ON -DPYTHON_EXECUTABLE="/usr/bin/python3" -DBUILD_FEM_NETGEN=ON ..
```

ceremcem, 2020-11-28:
> Why not just building using a ubuntu 18.04 LXC image

There were couple of subjective reasons, like I'm very familiar with Debian, I use BTRFS so a new LXC container does not occupy any space on my harddisk and any script written for Ubuntu should have work on Debian without any modification or with a little change. But I wouldn't require you to support also Debian for those subjective reasons, it's unnecessary. The compilation process is finished just now on Ubuntu Bionic and I filed the issue at https://github.com/realthunder/FreeCADMakeImage/issues/5.

ceremcem, 2020-11-30:
Freshly compiled FreeCAD works well. There is no such error occurred anymore.

closed 2020-11-30 by ceremcem