realthunder/FreeCAD_assembly3 #353: Wrong scale UI freecad

AcumenDev, 2020-11-27:
Hello, i have two monitors, 
```
user@host:/$ xrandr
Screen 0: minimum 320 x 200, current 4480 x 1440, maximum 8192 x 8192
XWAYLAND0 connected 1920x1080+2560+0 (normal left inverted right x axis y axis) 160mm x 90mm
   1920x1080     59.96*+
XWAYLAND1 connected 2560x1440+0+0 (normal left inverted right x axis y axis) 600mm x 340mm
   2560x1440     59.91*+

 
```
i don't use diferrent scale for each monitors.

My problem is wrong scale UI elements in freecad on large monitor all very small, on small monitor all very big.
i can't fix it.
monitor 2560x1440: 
[image]
monitor  1920x1080:
[image]

Original last build freecad(https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre) work is normaly.

i use this version 
OS: Debian GNU/Linux 10 (buster) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.7.8
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2020-11-28:
You are using an older release. It is more stable but lags quite far behind the upstream. Can you please try my release with `Daily` in its name? I just made a release today.

AcumenDev, 2020-11-28:
> You are using an older release. It is more stable but lags quite far behind the upstream. Can you please try my release with `Daily` in its name? I just made a release today.

Thanks for answer, but I don't find where dowdload Daily release 
I download this version from:  https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11
What am I doing wrong?

[image]

realthunder, 2020-11-29:
Yeah, I pull out the update because I found a problem, but didn't get it fixed in time. Anyway, I have uploaded 11.29 release. Please try.

AcumenDev, 2020-11-29:
thank you! daily build works well

closed 2020-11-29 by AcumenDev