realthunder/FreeCAD_assembly3 #354: losing the contents of toolbars

snelweg, 2020-11-27:
FreeCAD keeps losing the contents of the toolbars on screen. When I now start up FC most toolbars are empty. When I fill them with View->Toolbars they are there and work. But when I restart FC the toolbars are gone again. I thought wanted behaviour was that FC saves toolbar content modifications.
This not saving used to happen a lot a while ago, looks like this bug is back again. I tried emptying the .freecad folder and do all the settings again, this works for a little while and then FC stops saving again/
What can I do to get the saving of toolbars working again?

`OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.11.16.22739 +2766 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 6730ad3b3bd841c2745ec98bf953c523dea86913
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)`

jhoogeboom, 2020-12-23:
I have the same issue, what I found is that opening a file from an earlier release of realthunders asm3 triggers this behaviour. Also noticed this text in the release notes:

* Use different configuration file than upstream (link.system.cfg, link.user.cfg)

When using the latest release, creating a new file, I don't have any problems. I had to merge the old file with a new file to be able to prevent this losing the toolbars from happening.

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.17.23076 +3024 (Git)
Build type: Release
Branch: LinkStage3
Hash: 560eb2c4645cd548b6f718c645c7af70384ff54c
Python version: 3.7.9
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/Belgium (en_BE)
```

snelweg, 2021-01-30:
Still having this very anoying problem.
I didn't downgrade, only upgraded FC linkstage3 version. I didn't open a file made or edited by a previous version.

This is getting to the point that I keep my pc running 24 hours a day, just so I do not lose the toolbars. Unless FC crashes ofcourse, then mostly the toolbars are still there (but not all of them).
Not using any toolbar add-ons, they didn't do any good anyway.

What do I need to send so this problem can be solved for good?

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0129.23662 +3166 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: bd2d90d312b3519573a323f97d7544d82f0e314d
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-01-30:
I am aware of this problem, but can never reliably reproduce it. Once this occured, please send me your `link.user.cfg` configuration file. Not sure if it helps, though.

snelweg, 2021-01-30:
>please send me your `link.user.cfg` configuration file. 

It occurs every time.

`<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>
  <FCParamGroup Name="Root">
    <FCParamGroup Name="BaseApp">
      <FCParamGroup Name="Preferences">
        <FCParamGroup Name="General">
          <FCBool Name="SaveUserParameter" Value="1"/>
          <FCText Name="LastModule">ArchWorkbench</FCText>
          <FCText Name="FileOpenSavePath">/home/xx/Nextcloud/Hoogenweg/CV-hok</FCText>
          <FCText Name="AutoloadModule">PartDesignWorkbench</FCText>
          <FCBool Name="ShowSplasher" Value="1"/>
          <FCBool Name="PythonWordWrap" Value="1"/>
          <FCInt Name="ToolbarIconSize" Value="24"/>
          <FCText Name="FileImportFilter">All files (*.*)</FCText>
          <FCBool Name="ConfirmAll" Value="1"/>
          <FCText Name="OffscreenImageFormat">png</FCText>
          <FCInt Name="OffscreenImageBackground" Value="3"/>
          <FCText Name="FileExportFilter">STL Mesh (*.stl *.ast)</FCText>
          <FCText Name="PreferenceGeometry">1470 114 812 775</FCText>
        </FCParamGroup>
        <FCParamGroup Name="Macro">
          <FCBool Name="LocalEnvironment" Value="1"/>
          <FCText Name="MacroPath">/home/xx/.FreeCAD/Macro</FCText>
          <FCBool Name="RecordGui" Value="1"/>
          <FCBool Name="GuiAsComment" Value="1"/>
          <FCBool Name="ScriptToPyConsole" Value="1"/>
          <FCBool Name="ScriptToFile" Value="0"/>
          <FCText Name="ScriptFile">FullScript.FCScript</FCText>
        </FCParamGroup>
        <FCParamGroup Name="Mod">
          <FCParamGroup Name="Import">
            <FCParamGroup Name="hSTEP">
              <FCBool Name="ReadShapeCompoundMode" Value="0"/>
            </FCParamGroup>
            <FCBool Name="UseAppPart" Value="1"/>
            <FCBool Name="UseLegacyImporter" Value="0"/>
            <FCBool Name="UseBaseName" Value="1"/>
            <FCBool Name="ImportHiddenObject" Value="1"/>
            <FCBool Name="ExportHiddenObject" Value="1"/>
            <FCBool Name="ExportKeepPlacement" Value="1"/>
            <FCBool Name="ReduceObjects" Value="0"/>
            <FCBool Name="ShowProgress" Value="1"/>
            <FCBool Name="ExpandCompound" Value="1"/>
            <FCBool Name="ExportLegacy" Value="0"/>
            <FCInt Name="ImportMode" Value="0"/>
          </FCParamGroup>
          <FCParamGroup Name="OpenSCAD"/>
          <FCParamGroup Name="Part">
            <FCParamGroup Name="Boolean">
              <FCBool Name="CheckModel" Value="0"/>
              <FCBool Name="RefineModel" Value="0"/>
            </FCParamGroup>
            <FCParamGroup Name="IGES">
              <FCInt Name="Unit" Value="0"/>
              <FCBool Name="BrepMode" Value="0"/>
              <FCBool Name="SkipBlankEntities" Value="1"/>
              <FCText Name="Company"/>
              <FCText Name="Author"/>
            </FCParamGroup>
            <FCParamGroup Name="General">
              <FCInt Name="WriteSurfaceCurveMode" Value="1"/>
            </FCParamGroup>
            <FCParamGroup Name="STEP">
              <FCInt Name="Unit" Value="0"/>
              <FCText Name="Scheme">AP214IS</FCText>
              <FCText Name="Company"/>
[... 2152 lines cut ...]
            <FCBool Name="Saved" Value="1"/>
            <FCText Name="Top">Break,View,Measure,Break,Custom,Part Design Helper</FCText>
            <FCText Name="Left">Tabs</FCText>
            <FCText Name="Right"/>
            <FCText Name="Bottom"/>
          </FCParamGroup>
          <FCParamGroup Name="MeshWorkbench"/>
          <FCParamGroup Name="PovRayRendering"/>
        </FCParamGroup>
        
        
        
        <FCParamGroup Name="System"/></FCParamGroup>
    </FCParamGroup>
    <FCParamGroup Name="Plugins">
      <FCParamGroup Name="addonsRepository">
        <FCBool Name="readWarning" Value="1"/>
      </FCParamGroup>
      <FCParamGroup Name="ModernUI">
        <FCText Name="destination">/home/xx/.FreeCAD//Mod/ModernUI</FCText>
      </FCParamGroup>
    </FCParamGroup></FCParamGroup>

</FCParameters>
`

snelweg, 2021-01-30:
This is really weird. 

Had opened the link.user.cfg file in VSCode to replace my user name so I could send it to you. After sending I examined the file just being curious.
I noticed that most of the `<FCParamGroup Name="Toolbars">` had a value of '0', so I changed some of them to '1' with FC not running. Saved the link.user.cfg file to original place.
After starting FC all the toolbar entries which I changed the values to '1' worked and were present. Looks good.
But when I returned to VSCode, it automatically reloaded the link.user.cfg file because it was changed on disk, most of the values were switched back '0' without me doing anything.
So even without saving FC after successful startup saves the ToolBars values to '0'.

Maybe that is a clue to what is happening?

jhoogeboom, 2021-01-30:
Yeah it wasn't really solved here either, it does still seem to disable toolbars here and there. I do keep my `link.user.cfg` under repository control now.

ceremcem, 2021-01-31:
I thought I had been experiencing the same issue but I can't notice lately. My version is: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.26832 (Git)
Build type: Release
Branch: LinkStage3
Hash: d7b79274aab23bef2132b1094f222fd34104ffb8
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

Our difference is:
```
d7b79274aa (origin/LinkStage3, origin/HEAD) Gui: improve checked menu for toolbar and dock window
b17565e191 Sketch: fix sketch mirgration problem (again)
070f318f17 PartDesign: add tooltip when dropping on SubShapeBinder
0382961066 Gui: improve tree view tooltip show/hide behavior
bd2d90d312 Spreadsheet: allow middle underscore in cell alias
```

snelweg, 2021-01-31:
> I thought I had been experiencing the same issue but I can't notice lately. My version is:

Switched over to 2021.0130.23662 +3170 (Git) AppImage and it looks like it is fixed now. There is also another way added to pick multiple toolbars at once, much easier now with checkboxes (will save a lot of time if this turns out to not be fixed).

[image]

>OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0130.23662 +3170 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: d7b79274aab23bef2132b1094f222fd34104ffb8
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-02-04:
Can you please try 0202 release and see if this problem is fixed?

snelweg, 2021-02-04:
Looks like it is fixed since 2021.0130.23662 +3170 (Git) AppImage
Thanks again for making this the best FreeCAD there is.

closed 2021-02-04 by snelweg

snelweg, 2021-02-06:
This was fixed (for me) after 2021.0130.23662 +3170 .
Using the 2021.0205.23662 +3201 version now and it is still fixed.

>OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)