realthunder/FreeCAD_assembly3 #357: Linux AppImage Crash on start

JGReyes, 2020-11-30:
First of all thanks for your work. I have used this versión before without problem:

FreeCAD-asm3-Conda-Py3-Qt5-20200818-glibc2.12-x86_64.AppImage

OS: Linux Debian Testing (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.7.8
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0

But today i downloaded file: FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201129-glibc2.12-x86_64.AppImage and when start the report view show this:

`14:04:26  PySide2/__init__.py: Unable to import shiboken2 from /home/jose/.FreeCAD/Mod/workfeature, /home/jose/.FreeCAD/Mod/ThreadProfile, /home/jose/.FreeCAD/Mod/3D_Printing_Tools, /home/jose/.FreeCAD/Mod/PieMenu, /home/jose/.FreeCAD/Mod/kicadStepUpMod, /home/jose/.FreeCAD/Mod/ExplodedAssembly, /home/jose/.FreeCAD/Mod/Assembly4, /home/jose/.FreeCAD/Mod/BOLTSFC, /home/jose/.FreeCAD/Mod/Defeaturing, /home/jose/.FreeCAD/Mod/fasteners, /home/jose/.FreeCAD/Mod/Manipulator, /home/jose/.FreeCAD/Mod/parts_library, /home/jose/.FreeCAD/Mod/symbols_library, /home/jose/.FreeCAD/Mod/FCGear, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/asm3, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/appimage_updater, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Web, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Tux, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Test, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/TechDraw, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Surface, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Start, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Spreadsheet, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Sketcher, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Show, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Robot, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/ReverseEngineering, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Raytracing, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Points, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Path, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/PartDesign, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Part, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/OpenSCAD, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/MeshPart, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Mesh, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Measure, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Material, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Inspection, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Import, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Image, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Idf, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Fem, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Drawing, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Draft, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Complete, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Arch, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/AddonManager, /tmp/.mount_FreeCA3Sbv0x/usr/Mod, /tmp/.mount_FreeCA3Sbv0x/usr/lib, /tmp/.mount_FreeCA3Sbv0x/usr/lib64, /tmp/.mount_FreeCA3Sbv0x/usr/Ext, /tmp/.mount_FreeCA3Sbv0x/usr/bin, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python38.zip, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8/lib-dynload, /home/jose/.local/lib/python3.8/site-packages, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8/site-packages, /home/jose/.FreeCAD/Macro, /tmp/.mount_FreeCA3Sbv0x/usr/Macro
14:04:26  During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCA3Sbv0x/usr/Mod/AddonManager/InitGui.py
14:04:26  Please look into the log file for further information
14:04:26  PySide2/__init__.py: Unable to import shiboken2 from /home/jose/.FreeCAD/Mod/workfeature, /home/jose/.FreeCAD/Mod/ThreadProfile, /home/jose/.FreeCAD/Mod/3D_Printing_Tools, /home/jose/.FreeCAD/Mod/PieMenu, /home/jose/.FreeCAD/Mod/kicadStepUpMod, /home/jose/.FreeCAD/Mod/ExplodedAssembly, /home/jose/.FreeCAD/Mod/Assembly4, /home/jose/.FreeCAD/Mod/BOLTSFC, /home/jose/.FreeCAD/Mod/Defeaturing, /home/jose/.FreeCAD/Mod/fasteners, /home/jose/.FreeCAD/Mod/Manipulator, /home/jose/.FreeCAD/Mod/parts_library, /home/jose/.FreeCAD/Mod/symbols_library, /home/jose/.FreeCAD/Mod/FCGear, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/asm3, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/appimage_updater, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Web, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Tux, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Test, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/TechDraw, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Surface, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Start, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Spreadsheet, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Sketcher, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Show, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Robot, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/ReverseEngineering, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Raytracing, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Points, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Path, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/PartDesign, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Part, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/OpenSCAD, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/MeshPart, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Mesh, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Measure, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Material, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Inspection, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Import, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Image, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Idf, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Fem, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Drawing, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Draft, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Complete, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Arch, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/AddonManager, /tmp/.mount_FreeCA3Sbv0x/usr/Mod, /tmp/.mount_FreeCA3Sbv0x/usr/lib, /tmp/.mount_FreeCA3Sbv0x/usr/lib64, /tmp/.mount_FreeCA3Sbv0x/usr/Ext, /tmp/.mount_FreeCA3Sbv0x/usr/bin, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python38.zip, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8/lib-dynload, /home/jose/.local/lib/python3.8/site-packages, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8/site-packages, /home/jose/.FreeCAD/Macro, /tmp/.mount_FreeCA3Sbv0x/usr/Macro
14:04:26  During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Arch/InitGui.py
14:04:26  Please look into the log file for further information
14:04:27  PySide2/__init__.py: Unable to import shiboken2 from /home/jose/.FreeCAD/Mod/workfeature, /home/jose/.FreeCAD/Mod/ThreadProfile, /home/jose/.FreeCAD/Mod/3D_Printing_Tools, /home/jose/.FreeCAD/Mod/PieMenu, /home/jose/.FreeCAD/Mod/kicadStepUpMod, /home/jose/.FreeCAD/Mod/ExplodedAssembly, /home/jose/.FreeCAD/Mod/Assembly4, /home/jose/.FreeCAD/Mod/BOLTSFC, /home/jose/.FreeCAD/Mod/Defeaturing, /home/jose/.FreeCAD/Mod/fasteners, /home/jose/.FreeCAD/Mod/Manipulator, /home/jose/.FreeCAD/Mod/parts_library, /home/jose/.FreeCAD/Mod/symbols_library, /home/jose/.FreeCAD/Mod/FCGear, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/asm3, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/appimage_updater, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Web, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Tux, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Test, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/TechDraw, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Surface, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Start, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Spreadsheet, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Sketcher, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Show, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Robot, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/ReverseEngineering, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Raytracing, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Points, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Path, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/PartDesign, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Part, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/OpenSCAD, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/MeshPart, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Mesh, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Measure, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Material, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Inspection, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Import, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Image, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Idf, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Fem, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Drawing, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Draft, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Complete, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/Arch, /tmp/.mount_FreeCA3Sbv0x/usr/Mod/AddonManager, /tmp/.mount_FreeCA3Sbv0x/usr/Mod, /tmp/.mount_FreeCA3Sbv0x/usr/lib, /tmp/.mount_FreeCA3Sbv0x/usr/lib64, /tmp/.mount_FreeCA3Sbv0x/usr/Ext, /tmp/.mount_FreeCA3Sbv0x/usr/bin, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python38.zip, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8/lib-dynload, /home/jose/.local/lib/python3.8/site-packages, /tmp/.mount_FreeCA3Sbv0x/usr/lib/python3.8/site-packages, /home/jose/.FreeCAD/Macro, /tmp/.mount_FreeCA3Sbv0x/usr/Macro
[... cut ...]
<class 'SystemError'>: <method-wrapper '__get__' of getset_descriptor object at 0x7f68b6214dc0> returned a result with an error set
14:11:49  <method-wrapper '__get__' of getset_descriptor object at 0x7f68b6214dc0> returned a result with an error set`

OS: Linux Debian Testing (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.11.29.23076 +2859 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: a491be7f04516e65647d77ceb2547fbd0adac1df
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Spanish/Spain (es_ES)

Thanks

realthunder, 2020-11-30:
Can please check your environment variable `LD_LIBRARY_PATH` value. Just type the following command in your terminal,

```
echo $LD_LIBRARY_PATH
```

If it has any value. Please clear it in your terminal before starting the AppImage and try again.

```
export LD_LIBRARY_PATH=
./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201129-glibc2.12-x86_64.AppImage
```

JGReyes, 2020-11-30:
Thanks for the quick response. I have nothing in $LD_LIBRARY_PATH, noways I run the sugested commands:

> export LD_LIBRARY_PATH=
./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201129-glibc2.12-x86_64.AppImage

and I get in the console the following:

`./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201129-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCA2tYlqp
FreeCAD 0.19, Libs: 2020.11.29R23076 +2859 (Git)
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
<Command> Command.cpp(1901): duplicate command Std_TreePreSelection
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCA2tYlqp/usr/Mod/AddonManager/InitGui.py
Please look into the log file for further information
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCA2tYlqp/usr/Mod/Arch/InitGui.py
Please look into the log file for further information
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCA2tYlqp/usr/Mod/Draft/InitGui.py
Please look into the log file for further information
During initialization the error "name '_init_pyside_extension' is not defined" occurred in /tmp/.mount_FreeCA2tYlqp/usr/Mod/Tux/InitGui.py
Please look into the log file for further information
During initialization the error "name '_init_pyside_extension' is not defined" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 156, in InitApplications
  File "/tmp/.mount_FreeCA2tYlqp/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/.mount_FreeCA2tYlqp/usr/Mod/asm3/freecad/asm3/init_gui.py", line 3, in <module>
    from .utils import mainlogger as logger
  File "/tmp/.mount_FreeCA2tYlqp/usr/Mod/asm3/freecad/asm3/utils.py", line 10, in <module>
    import FreeCAD, FreeCADGui, Part, Draft
  File "/tmp/.mount_FreeCA2tYlqp/usr/Mod/Draft/Draft.py", line 38, in <module>
    import Draft_rc
  File "/tmp/.mount_FreeCA2tYlqp/usr/Mod/Draft/Draft_rc.py", line 5, in <module>
    from PySide2 import QtCore
  File "/home/jose/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "/home/jose/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 58, in _setupQtDirectories
    import shiboken2
  File "/home/jose/.local/lib/python3.8/site-packages/shiboken2/__init__.py", line 30, in <module>
    _init_pyside_extension()
NameError: name '_init_pyside_extension' is not defined
--------------------------------------------------------------------------------
During initialization the error "name '_init_pyside_extension' is not defined" occurred in freecad.appimage
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 156, in InitApplications
  File "/tmp/.mount_FreeCA2tYlqp/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/.mount_FreeCA2tYlqp/usr/Mod/appimage_updater/freecad/appimage/init_gui.py", line 6, in <module>
    from . update import AppImageUpdaterDialog
  File "/tmp/.mount_FreeCA2tYlqp/usr/Mod/appimage_updater/freecad/appimage/update.py", line 4, in <module>
    from PySide2 import QtWidgets, QtGui, QtCore
  File "/home/jose/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "/home/jose/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 58, in _setupQtDirectories
    import shiboken2
  File "/home/jose/.local/lib/python3.8/site-packages/shiboken2/__init__.py", line 30, in <module>
    _init_pyside_extension()
[... 13 lines cut ...]
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/.mount_FreeCA2tYlqp/usr/lib/python3.8/site-packages/freecad/pip/init_gui.py", line 2, in <module>
    from PySide import QtCore, QtGui
  File "/tmp/.mount_FreeCA2tYlqp/usr/Ext/PySide/__init__.py", line 2, in <module>
    from PySide2 import __version__
  File "/home/jose/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "/home/jose/.local/lib/python3.8/site-packages/PySide2/__init__.py", line 58, in _setupQtDirectories
    import shiboken2
  File "/home/jose/.local/lib/python3.8/site-packages/shiboken2/__init__.py", line 30, in <module>
    _init_pyside_extension()
[... cut ...]
  File "/tmp/.mount_FreeCA2tYlqp/usr/lib/python3.8/site-packages/six.py", line 320, in <module>
    moves = _MovedItems(__name__ + ".moves")
  File "/tmp/.mount_FreeCA2tYlqp/usr/lib/python3.8/site-packages/six.py", line 128, in __init__
    self.__doc__ = self.__class__.__doc__
<class 'SystemError'>: <method-wrapper '__get__' of getset_descriptor object at 0x7f094e4e3dc0> returned a result with an error set
<method-wrapper '__get__' of getset_descriptor object at 0x7f094e4e3dc0> returned a result with an error set
`

JGReyes, 2020-12-03:
I got this issue solved finally, after the same error with the new version: FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201202-glibc2.12-x86_64.AppImage, I read carefully the log and find multiple entries of pyside2 and shiboken2 pointing into my local python3.8 site-package directory like this:

`File "/home/jose/.local/lib/python3.8/site-packages/PySide2/init.py", line 58, in _setupQtDirectories
import shiboken2
File "/home/jose/.local/lib/python3.8/site-packages/shiboken2/init.py", line 30, in
_init_pyside_extension()`

Is like the Appimage don't use his internal python3.8. I try to execute from a console with conda enviroment and python 3.7 and I get the same error on python3.8.

Finally because I don't use any app with pyside2 in python3.8, I uninstall both: pytside2 and shiboken2 with:

> pip3 uninstall pyside2

> pip3 uninstall shiboken2

and Reportview is no show any errors.

Regards

closed 2020-12-03 by JGReyes