realthunder/FreeCAD_assembly3 #363: No solver backend found

ceremcem, 2020-12-01:
I rebuilt everything from scratch but I think [Assembly3 Build Instructions](https://github.com/realthunder/FreeCAD_assembly3/wiki/Build-Instruction) are out of date. I can't install Assembly3 WB at all if I follow the instructions (which says `git clone https://github.com/realthunder/FreeCAD_assembly3  $build_dir/Ext/freecad/asm3`). 

If I clone Assembly3 WB into `~/.FreeCAD/Mod`, like `git clone  https://github.com/realthunder/FreeCAD_assembly3 ~/.FreeCAD/Mod/asm3`, then Assembly3 WB appears within the "workbenches" dropdown, but the following warning is displayed: 

```
19:18:58  3.521046 <asm3.main> init_gui.py(14): no solver backend found
```

There is no `slvs` submodule available in Assembly3 repo and if I'm not wrong, we were installing `py-slvs` over `pip`: 
```
$ sudo pip install py-slvs
Collecting py-slvs
  Using cached https://files.pythonhosted.org/packages/62/81/0e0f35959e228960d3a5f6784efcca6be3016ee7ab72b50d5831a04b9e66/py_slvs-1.0.0-cp27-cp27mu-manylinux1_x86_64.whl
Installing collected packages: py-slvs
Successfully installed py-slvs-1.0.0
```

So I think `py-slvs` is installed. However the warning message stays.

How should I proceed?

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25939 (Git)
Build type: Release
Branch: LinkStage3
Hash: 09625d1600eedb95404c879c4e9a0b0df0af9525
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

```
aea@fc4:~/.FreeCAD/Mod/asm3$ git log --oneline                                                                                                                          
fe6a792 (HEAD -> master, origin/master, origin/HEAD) assembly: fix workplane orientation   
```

ceremcem, 2020-12-02:
Fixed by compiling `py_slvs` with Python3 flag: [`install-slvs.sh`](https://github.com/ceremcem/build-freecad-asm3/blob/663e32d1e7b07028ad70686a82ec1483d19c685c/install-slvs.sh)

```bash
asm3=path/to/fc-build/Mod/asm3/freecad/asm3

git clone https://github.com/realthunder/solvespace.git $asm3/slvs
cd $asm3/slvs
git submodule update --init extlib/libdxfrw
mkdir -p build && cd build
cmake -DBUILD_PYTHON=1 -DPYTHON_EXECUTABLE:FILEPATH='/usr/bin/python3' ..
make _slvs
py_slvs_dir="$asm3/py3_slvs"
mkdir -p $py_slvs_dir
cp $asm3/slvs/build/src/swig/python/{slvs.py,_slvs.so} $py_slvs_dir
touch $py_slvs_dir/__init__.py
```

closed 2020-12-02 by ceremcem

lorinc, 2023-12-08:
Just [installed this branch from snap](https://snapcraft.io/install/freecad-realthunder/pop on pop_os), got this error. I really wish I would not have to compile anything for FreeCAD to work.