realthunder/FreeCAD_assembly3 #365: One edge from a sketch is not highlighted after mouse selection

Rabbit-sk, 2020-12-03:
Dear Realthunder,

There is allways one edge in a sketch, which is not higlighted after selecting it with mouse. I highlighted the particular edges in Python console:
[image]
[image]

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.02.23076 +2874 (Git)
Build type: Release
Branch: LinkStage3
Hash: f45215c049a756a8d5f072535eafbb308113f2b1
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Slovak/Slovakia (sk_SK)

Regards

realthunder, 2020-12-03:
This problem has been fixed https://github.com/realthunder/FreeCAD/commit/55073409dea8cb380db190fe906b5c3a28c44b07

I'll make a new release soon.

Rabbit-sk, 2020-12-03:
Later I find that also "Pick geometries G, G" -> "Wire" have the same outcome:
[image]
[image]

Rabbit-sk, 2020-12-03:
> This problem has been fixed realthunder/FreeCAD@5507340
Thanks

closed 2020-12-03 by Rabbit-sk