realthunder/FreeCAD_assembly3 #366: Context menu in 3D View is partly transparent

Rabbit-sk, 2020-12-03:
Dear Realthunder,

Context menu in 3D View is partly transparent. Because of this, reading of items in the context menu is uncomfortable:
[image]

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.02.23076 +2874 (Git)
Build type: Release
Branch: LinkStage3
Hash: f45215c049a756a8d5f072535eafbb308113f2b1
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Slovak/Slovakia (sk_SK)

Regards

realthunder, 2020-12-03:
You can change the the menu stylesheet in Preference -> General. Choose `Default` will use the default menu. You might want to experiment with different stylesheet combinations for a better look. May I ask what stylesheet are you using? I didn't have this problem.

Rabbit-sk, 2020-12-03:
Appearance of context menu with default settings in current daily release (0.19.23141) of official branch:
[image]
Colors are corresponding with the default style sheet and it is not transparent. 

> You might want to experiment with different stylesheet combinations for a better look. 

Although I solved this for me by setting "View menu style sheet" to "Light" (and get used to a little transparency), the default setting should have proper look.

closed 2020-12-04 by realthunder