realthunder/FreeCAD_assembly3 #368: Big Sur start fails but it worked

sigxcpu76, 2020-12-03:
I know that FreeCAD needs to be recompiled (https://forum.freecadweb.org/viewtopic.php?f=4&t=51345&start=40) for Big Sur, otherwise it doesn't start.
The funny thing is that 2 or 3 versions back FreeCAD Link was able to start. I've downloaded 2020.12.03 today and it fails to start. It eats one CPU core and doesn't show even the splash.

realthunder, 2020-12-04:
Can you find out exactly which version can run?

sigxcpu76, 2020-12-04:
I would've mentioned already but unfortunately I've overwritten it with 3.12 upgrade :(

realthunder, 2020-12-04:
Maybe you can try [here](https://drive.google.com/drive/folders/19eIWuQGtKR3GMQA66xBstoQ43b_5LFGT?usp=sharing)?

sigxcpu76, 2020-12-04:
2020.11.16 works (I lied, found the dmg in my Downloads folder). 2020.11.26 does not work. 202.11.29 does not work.

realthunder, 2020-12-06:
Can you please try [this](https://drive.google.com/file/d/1sPt9brTvdU2MmWqCzGHCsd8O6XW62rtA/view?usp=sharing) image and see if it works?

sigxcpu76, 2020-12-06:
Sure. I will do it tomorrow because I don't have the laptop with me atm.

sigxcpu76, 2020-12-13:
@realthunder if there is nothing more to add you/I can close this to avoid open issues pollution.

closed 2020-12-13 by realthunder