realthunder/FreeCAD_assembly3 #369: Please bundle daily releases under different app name

akamensky, 2020-12-06:
Hey, me again. I have a suggestion -- please make releases under different app name (and different config directory), to allow me having upstream FreeCAD and this fork running in parallel. Right now I have to download your daily builds, try them, then overwrite it back with upstream builds. That is a bit exhausting. That is not just app name. I am on macos, so I can just rename "FreeCAD.app" to whatever, but it uses same config directory, which makes it much more complicated.

closed 2020-12-07 by realthunder

realthunder, 2020-12-07:
May I ask what setting in particular you find that's incompatible with upstream? Is it those UI settings? The referenced commit will make my release image to save those `Preference` settings and the parameters (accessible through `Edit parameters`) in different files (specifically, `system.cfg` vs. `link.system.cfg` and `user.cfg` vs. `link.user.cfg`).

All other user configurations are still shared, such as, Macro directory, user installed modules, etc. Will that be enough?

akamensky, 2020-12-07:
@realthunder I am not sure whether that will be enough. There seems to be some incompatibilities as I tried initially open Link daily builds with upstream config directory (so not changing anything) and I get a lot of weird issues like files failing to open, broken UI etc. I did change a bunch of settings in upstream installation, so there may be some incompatibilities for those?

What I ended up doing is -- backup upstream config directory and delete them, open Link daily release, try it out, find a bunch of issues (all reported here already) and then delete config directory (create by Link) and restore the one backed up for upstream.

That appears the only way currently to make sure there is no issues caused by incompatible configs.

If they share directory there still may be some issues, for example any previously installed mods may be not compatible? (I currently only have Fasteners workbench and A2plus installed). Which is why I think just using different configuration directory name would be more beneficial here, as I could just try the daily builds without worrying about accidentally overwriting my upstream configs.