realthunder/FreeCAD_assembly3 #370: Errors trying to open upstream created files

akamensky, 2020-12-06:
Files created using upstream FreeCAD releases are failing to open:

```
12:56:57  During initialization the error "'ascii' codec can't decode byte 0xc3 in position 108: ordinal not in range(128)" occurred in /Applications/FreeCAD Link.app/Contents/Resources/Mod/Web/InitGui.py
12:56:57  Please look into the log file for further information
12:56:57  During initialization the error "'ascii' codec can't decode byte 0xc3 in position 210: ordinal not in range(128)" occurred in /Applications/FreeCAD Link.app/Contents/Resources/Mod/Surface/InitGui.py
12:56:57  Please look into the log file for further information
12:57:09  Cannot read file:  ... redacted ...
12:57:09  Cannot read file:  ... redacted ...
12:57:09  Cannot read file:  ... redacted ...
12:57:09  Cannot read file:  ... redacted ...
12:57:09  Cannot read file:  ... redacted ...
12:57:09  Cannot read file:  ... redacted ...
12:57:09  Cannot read file:  ... redacted ...
12:57:21  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: ios_base::clear: unspecified iostream_category error
```

ceremcem, 2020-12-06:
It might be helpful to add the failing example file here.

akamensky, 2020-12-06:
Version is `Daily-OSX-Conda-Py3-Qt5-20201203` as downloaded from releases page.

Files -- just any. I.e. this one of MGN12H block: 
[6. MGN12H block.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5648883/6.MGN12H.block.FCStd.zip)

sigxcpu76, 2020-12-06:
I've opened this file (with some warnings) on 12.03 release (on Linux).

ceremcem, 2020-12-07:
I got this response while opening above file: 

```
07:23:02  Fatal Error at file /home/aea/tmp/6.MGN12H.block.FCStd, line 1, char 1
07:23:02  Exception message is: 
07:23:02  invalid document structure
07:23:02  <Exception> Error reading project file: /home/aea/tmp/6.MGN12H.block.FCStd
07:23:02  <Exception> Unknown file exception happened
```

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.25981 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0058fa7c2ab22e32e069576046d005835da15ffa
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

akamensky, 2020-12-07:
Thanks @ceremcem. It does look inconsistent to me, I managed to open above file in this fork, but then I faced #372.

ceremcem, 2020-12-07:
Regardless of `~/.FreeCAD` directory contents, it gives the very same error.

realthunder, 2020-12-07:
I can't reproduce this. Do you have trouble opening any other file besides this one?

akamensky, 2020-12-07:
It was multiple files for me, this one linked because it is downloaded from online. There were a few others that also failed that I created in recent 0.19 upstream release (22xxx, so not that old).

ceremcem, 2020-12-07:
Ah! Pardon me. I saw the extension `.FCStd.zip`, so I just removed the `.zip` at the end, instead of unzipping it (me and some other  guys were just adding `.zip` to workaround Github's upload restrictions). It's now opened. 

As I don't have Fasteners WB, it threw that error: 

```
08:35:14  <string>(1)<class 'ModuleNotFoundError'>: No module named 'FastenersCmd'
```

There are lots of warnings, but it's okay: 

```
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Body
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Sketch
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Sketch001
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Sketch002
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pocket
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pocket
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer001
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer001
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer002
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer002
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Sketch003
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Hole
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Hole
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Body001
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#CopySketch004
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad001
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad001
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer003
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer003
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Body002
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#CopySketch004001
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad002
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad002
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer004
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer004
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Body003
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#CopySketch004002
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad003
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad003
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer005
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer005
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Body004
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer006
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Chamfer006
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad004
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#Pad004
08:35:13  <PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: ___MGN12H_block#CopySketch004003
[... cut ...]
08:35:26  <App> GeoFeatureGroupExtension.cpp(574): ___MGN12H_block#Pad003.Profile(links to ___MGN12H_block#CopySketch004002) is out of scope:<global> vs. ___MGN12H_block#Body003
08:35:26  <App> GeoFeatureGroupExtension.cpp(574): ___MGN12H_block#Pad004.Profile(links to ___MGN12H_block#CopySketch004003) is out of scope:<global> vs. ___MGN12H_block#Body004
```

[image]

(Lessons learnt: We should never append a `.zip` postfix. Instead, we should add `.FCStd.jpeg` or some weird extension for that purpose.)

ceremcem, 2021-02-21:
@akamensky Is this issue resolved?

akamensky, 2021-02-25:
Haven't tested cuz I gave up on asm3. I could not use it after all.

ceremcem, 2021-02-25:
Wow, that's bad news. What was the main issue you have struggled?

akamensky, 2021-02-26:
Short version - overly complicated workflow for a simple use case and numerous issues that are considered minor, but made it quite difficult to accomplish my simple use case.

ceremcem, 2021-02-26:
Sorry to hear that. Let me know if I could assist you for a startup.

Alexey Kamenskiy <notifications@github.com>, 26 Şub 2021 Cum, 11:33
tarihinde şunu yazdı:

akamensky, 2021-02-26:
Thanks. I am already at n the middle of the work with Asm2+. Don't think I want to derail it to start everything from scratch.

luzpaz, 2021-06-24:
Closing ticket

closed 2021-06-24 by luzpaz