realthunder/FreeCAD_assembly3 #371: Is it possible to use system dialogs in this fork?

akamensky, 2020-12-06:
Upstream releases use system dialogs for open/save/etc. This fork uses something else, which does not remember any previous choices/locations and makes navigation so much more difficult.

Upstream dialogs:

[image]

This fork dialogs:

[image]

akamensky, 2020-12-06:
One problem with using non-standard dialog is that I am unable to use external drive at all. "/Volumes" is not showing up at all:

[image]

sigxcpu76, 2020-12-06:
Tools/Edit Parameters/Preferences/Dialog

Set `DontUseNativeDialog` to `false`. Add a boolean key if it is not there.

[image]

Rabbit-sk, 2020-12-06:
> Set `DontUseNativeDialog` to `false`. Add a boolean key if it is not there.

Is there a reason why it is not set like this by default?

akamensky, 2020-12-06:
@sigxcpu76 thanks, I will try it out once I am finished with current work for the day. I switched back to using upstream version due to multiple issues (a few issues I reported here today).

Agree with @Rabbit-sk. I think it should be using system one by default since the upstream version already does use that. I don't think there is any advantage to using non-system one in this instance.

closed 2020-12-06 by realthunder

realthunder, 2020-12-06:
I have change back to default use native file dialog on Windows and MacOSX, same as upstream. The reason for using Qt file dialog is that this is the only way (that I know of) to open and save project into directory, especially save to a new directory. For existing project in an uncompressed directory, you can select the `Document.xml` file in the native file dialog.

Anyhow, I have expose the file dialog setting as below. And you can temporarily override the setting by holding SHIFT key when open or save your project.

[image]

akamensky, 2020-12-06:
Not sure about Windows, on macos I've seen a couple of applications doing directory based save/load from native dialog. But this may be just macos specific since it supports defining directories as "file type" (i.e. `.app` directory to which applications are bundled).

sigxcpu76, 2020-12-06:
On Linux (at least) native file dialog does not add extensions automatically. Extensionless file name is also added to "recent files" which makes it impossible to open them from there. You need to put `.FCStd` by hand or `.stl` on exporting for 3d printing.

Rabbit-sk, 2020-12-07:
I apologize for reopening this topic. I have a suggestion (maybe it has no sense for Linux, I don't know). In File menu, there could be two new items, one: "Open uncompressed  project" (after Open... Ctrl+O), and second: "Save uncompressed project" (maybe after Save as...). These two could use Qt file dialog, and standard ones could use native file dialogs.
[image]