realthunder/FreeCAD_assembly3 #372: Upstream saved files are losing geometry on recompute (opening in this fork)

akamensky, 2020-12-06:
Using nightly releases opening (open-source copy of) MGN12H block model in upstream shows correct model:

[image]

Opening the same model in this fork produces invalid model (after forced recompute):

[image]

Model in question https://www.thingiverse.com/thing:4249218/files

realthunder, 2020-12-07:
With the above referenced commit, the behavior should now be the same as upstream. After recompute, you'll see the hole shrink, which may be caused by recent change in PartDesign hole feature in upstream. If you open the file in upstream, and mark the whole document for recompute, you'll see the same shrinking behavior. The reason for recompute in my image is for generating new topo names.

I'll make a new release soon.

luzpaz, 2021-02-20:
@realthunder can this ticket be closed ?

closed 2021-02-21 by realthunder