realthunder/FreeCAD_assembly3 #373: [path] cannot remove a job object from treee

snelweg, 2020-12-09:
`14:43:11  Traceback (most recent call last):
  File "C:\Users\woo_\Downloads\FC_BIN\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20201208\Mod\Path\PathScripts\PathJob.py", line 361, in execute
    obj.Path = obj.Operations.Path
<class 'AttributeError'>: 'NoneType' object has no attribute 'Path'
14:43:11  Recompute failed! Please check report view.
14:43:17  Traceback (most recent call last):
  File "C:\Users\woo_\Downloads\FC_BIN\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20201208\Mod\Path\PathScripts\PathJob.py", line 361, in execute
    obj.Path = obj.Operations.Path
<class 'AttributeError'>: 'NoneType' object has no attribute 'Path'
14:43:17  Recompute failed! Please check report view.
14:43:23  Traceback (most recent call last):
  File "C:\Users\woo_\Downloads\FC_BIN\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20201208\Mod\Path\PathScripts\PathJob.py", line 361, in execute
    obj.Path = obj.Operations.Path
<class 'AttributeError'>: 'NoneType' object has no attribute 'Path'
14:43:23  Recompute failed! Please check report view.
14:43:28  Traceback (most recent call last):
  File "C:\Users\woo_\Downloads\FC_BIN\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20201208\Mod\Path\PathScripts\PathJobGui.py", line 229, in onDelete
    self.obj.Proxy.onDelete(self.obj, arg2)
  File "C:\Users\woo_\Downloads\FC_BIN\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20201208\Mod\Path\PathScripts\PathJob.py", line 203, in onDelete
    PathLog.debug('taking down ops: %s' % [o.Name for o in self.allOperations()])
  File "C:\Users\woo_\Downloads\FC_BIN\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20201208\Mod\Path\PathScripts\PathJob.py", line 430, in allOperations
    for op in self.obj.Operations.Group:
<class 'AttributeError'>: 'NoneType' object has no attribute 'Group'`

`OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.08.23076 +2912 (Git)
Build type: Release
Branch: LinkStage3
Hash: 1eba27283118bae43f5264d38caf7451b848f49a
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Dutch/Netherlands (nl_NL)
`

ceremcem, 2021-02-21:
@snelweg Is it still valid?

snelweg, 2021-02-21:
@ceremcem As I am using the AppImage and there is no new version since my report, yes this is still valid.

luzpaz, 2021-02-21:
We need a windows user to test this

Rabbit-sk, 2021-02-21:
I use Windows and opened version 2020.12.08.23076 +2912 (Git) and tried to recreate this issue with some simple model, but deleting job was done without any warning or error. @snelweg could you provide us the project file?

snelweg, 2021-02-22:
@Rabbit-sk I do not have the file anymore.

closed 2021-02-22 by snelweg