realthunder/FreeCAD_assembly3 #376: Sketcher makes FC crash when I do a cancel on an operation

snelweg, 2020-12-10:
When I am in Sketch WB and cancel out on an operation FC mmediately crashes.

It is so bad that I can not use this release of FC anymore.

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.08.23076 +2912 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 1eba27283118bae43f5264d38caf7451b848f49a
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

Rabbit-sk, 2020-12-10:
Could you please provide more detailed information (steps how to recreate the issue, screenshot)? I tried canceling the geometry drawing, dimensioning the line, or editing the sketch, but I didn't get any errors. It is true that my PC runs Windows, but anyway more info should be provided to address this issue.

snelweg, 2020-12-10:
> steps how to recreate the issue, screenshot?

In the screenshot there is the situation where FC will crash when I choose <cancel>.
I pad a sketch and then use cancel to not go ahead, FC crashes.

[image]

I attached the FC file too for you convenience.

[cube_cutout.FCStd.txt](https://github.com/realthunder/FreeCAD_assembly3/files/5674633/cube_cutout.FCStd.txt)

Rabbit-sk, 2020-12-10:
I recreated this issue. Once it crasches, another time not, but last solid feature in list is left hiddden. Here is screenshot of Report view (View -> Panels -> Report view):
[image]

Although I'm not the developer, I posted my question so it would be easier for @realthunder to look at this issue.

Regards

snelweg, 2020-12-10:
> I recreated this issue. Once it crasches, another time not, 

It crashes consistently here.

>Here is screenshot of Report view (View -> Panels -> Report view):

I never see that, because when it crashes all panels are gone.

Rabbit-sk, 2020-12-10:
> I never see that, because when it crashes all panels are gone.

I usually set report view to be visible at screen when I try to recreate the issue. In my case if FC crashes it is not instattly, so I have a few seconds to make a screenshot.

closed 2020-12-11 by realthunder