realthunder/FreeCAD_assembly3 #377: Techdraw leaderline doesn't attach to the view

ngocminhdao88, 2020-12-12:
Hi,

Techdraw leaderline doesn't attach to the selected view or given position. It jumps to bottom left of the drawing page.

Here is the error i got from report view.
[image]

Here is what happens to me
[image]

I am using this version of FreeCAD
> OS: Ubuntu 20.04.1 LTS (ubuntu:GNOME/ubuntu)
> Word size of OS: 64-bit
> Word size of FreeCAD: 64-bit
> Version: 2020.12.08.23076 +2912 (Git) AppImage
> Build type: Release
> Branch: LinkStage3
> Hash: 1eba27283118bae43f5264d38caf7451b848f49a
> Python version: 3.8.6
> Qt version: 5.12.9
> Coin version: 4.0.0
> OCC version: 7.4.0
> Locale: English/United States (en_US)

realthunder, 2020-12-12:
Is it possible to attach your file here?

ngocminhdao88, 2020-12-12:
Here is it. It is just a simple cube for the sake of demonstration.

[leaderline-error.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5682847/leaderline-error.zip)

closed 2020-12-13 by realthunder