realthunder/FreeCAD_assembly3 #380: OBJ export contains only original linked object, not link instances

Rabbit-sk, 2020-12-15:
Dear Realthunder,

I made many copies of the objects using the link function. In OBJ export, however, there is only the original object and its "link copies" not. Creating a meshes in Mesh Design is out of the question as my scene contains hundreds of objects. How one could export all Link instances to OBJ format?

Regards

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.13.23076 +3015 (Git)
Build type: Release
Branch: LinkStage3
Hash: eb10f4011b36086ed70e8f667c70c10c6d2ecb3e
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

realthunder, 2020-12-17:
New image released with some fix in OBJ export/import. Note, there are two OBJ format support in FreeCAD. You probably want the `Wavefront OBJ format`.

Rabbit-sk, 2020-12-17:
I'm testing 2020-12-17 release now. I will report more about this issue with screenshots, as it seems that this fix did not fix everything.

Rabbit-sk, 2020-12-19:
Linked objects are now exported, but if Link object is made from Part with multiple bodies, in exported file they are grouped together as one object. This is not good for texturing purposes.
[image]
[image]
BTW which setting have effect to meshing accuracy of OBJ format? `Preferences - Import-Export - Mesh Formats - Export - Maximum mesh deviation` doesn't seem to have an effect.

Rabbit-sk, 2020-12-19:
Objects created as Links are now exported, but there is another problem that I reported here: #386

closed 2020-12-19 by Rabbit-sk