realthunder/FreeCAD_assembly3 #381: Collada *.dae export does not work

Rabbit-sk, 2020-12-15:
Dear Realthunder,

when I tried to export my model to the dae format, no file was created and this error was logged to the Report View:
`13:43:42  pycollada not found, collada support is disabled.`

Regards

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.13.23076 +3015 (Git)
Build type: Release
Branch: LinkStage3
Hash: eb10f4011b36086ed70e8f667c70c10c6d2ecb3e
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

realthunder, 2020-12-17:
New image released with pycollada pre-installed. BTW, as you are using Windows release, it is relatively easy to install your own Python modules. Just start a `cmd` prompt, `cd` to the `bin` directory, and run `python -m pip install <module_you_want>`.

Rabbit-sk, 2020-12-17:
Thank you for the information (at the page [Extra python modules – FreeCAD Documentation](https://wiki.freecadweb.org/Extra_python_modules#Windows_3), for pycollada on Windows there is only: `On Windows since 0.15 pycollada is included in both the FreeCAD release and developer builds so no additional steps are necessary.`). Thank you for pre-installing pycollada so it will work out of the box, also as in the official branch.

closed 2020-12-17 by Rabbit-sk