realthunder/FreeCAD_assembly3 #382: Unnamed objects in tree after saving from directory to file

johnsonm, 2020-12-16:
Thanks for all the work on your branch! I love the ability to save as directory in Git. It's wonderful to be able to use version control reasonably.

I am working on a project, currently using FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201214-glibc2.12-x86_64.AppImage

I've been saving to a directory in this Git repository: https://gitlab.com/mcdanlj/applianceshelf

In order to share with the FreeCAD forum, I saved as a FCStd file: https://forum.freecadweb.org/download/file.php?id=136506

I can still load successfully from the directory, but the FCStd file loads with an error and drops all the structure:
[image]

I've done this twice (first, I think, with FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201022-glibc2.12-x86_64.AppImage) and it apparently happened with the older version too, but it didn't occur to me to try reading it myself to reproduce the problem earlier.

It seems that this will reproduce at will by opening the ApplianceShelf directory (from the git repository) and then saving it to an ApplianceShelf.FCStd file

I'm relatively new to this, so I'm sure I've made some parts of the design tree ugly and might have made mistakes that could have precipitated this corruption.

Conversation on the FreeCAD forum where I asked for advice: https://forum.freecadweb.org/viewtopic.php?f=20&t=52152

johnsonm, 2020-12-16:
First, _everything_ I have done since the corrupted commit I can re-do without trouble. All I'm doing is arranging parts to cut on the router.

This is a design for flat-pack style assembly, where all the parts are 2.5D routed out of plywood sheet. So I have one assembly of all the parts together to make sure they fit, and other assemblies laid out to route out of plywood in jobs that are meant to fit the available space of my CNC router. The parts were designed in their final assembled orientation, not in their packed-to-cut orientation.

A key aspect is that one particular size (the depth of the middle shelf, its Y dimension) is driven from a driven dimension in a sketch in another part. The depth, Y dimension of the top shelf is driven parametrically in the first sketch for the side. That's the main real dependency between the parts. Other than that they are basically all driven from fixed parameters in the sheet.

That corrupted commit was the first commit in which I had to rotate a part to an orientation that would fit on the sheet. I expect that's why there is an Angle property. I found that merely creating a plane coincident constraint was insufficient to rotate pieces to fit, so I used the axial move to rotate to correct orientation as well as to move to approximate location, and then added plane coincident constraints with parametric offset (space for cutting bit) to align parts packed to cut.

I was absolutely experimenting with other ways of doing this and tried different constraints and using undo and deleting constraints and elements that were left broken after removing experimental constraints, so it's entirely possible this is one of those bugs that are hard to reproduce from a simple sequence of actions. (I'm still flailing around trying things to try to understand what works and what doesn't; this is an experiment and I'm not expert.) I definitely don't remember all the different things I tried in that process, though. I can certainly revert and make another branch and see if the problem recurs.

There might be two bugs here, though, since user jpg87 [reported difficulty](https://forum.freecadweb.org/viewtopic.php?f=20&t=52152&p=458084#p449018) even trying to load the earlier FCStd file that I had created from commit 48df43eafc037ed69137768ac304eff1f40d051c (I believe; I didn't think to tag it): https://gitlab.com/mcdanlj/applianceshelf/uploads/6d2b7d04d55ccb9507574fd604a954cd/ApplianceShelf.FCStd 

I don't have details on what jpg87 saw beyond the forum post, so I'm not sure how to report that bug usefully, if it is indeed a separate bug.

realthunder, 2020-12-16:
I can somehow reproduce this problem using the AppImage by simply recompute and save again your document with the good commit. But I can't reproduce it in my development build. Maybe some more obscured memory corruption is involved. I'll investigate more.

realthunder, 2020-12-17:
The problem should be fixed now. I have just made a new release. Please try it with your document with the good commit.

johnsonm, 2020-12-17:
I confirm that it is fixed in FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201217-glibc2.12-x86_64.AppImage

I checked out bad commit b9edcafb6a1b350bd46681fe0a29810653ecef43, opened it, saved as file, closed it, opened the file, and still got this:

[image]

Then I checked out prior good commit 7fb48e4532486b673c6423d8bd66c420f90d3b0c, opened, saved as file, closed, opened the file, and it opened successfully.

Thank you!

closed 2020-12-17 by johnsonm