realthunder/FreeCAD_assembly3 #387: Changing a named reference constraint crashes FreeCAD

papamat, 2020-12-21:
I can constantly recreate this issue following this procedure.

- New File
- New Body
- New Sketch
- Add an arbitrary line
- Add a constraint in (length) and make it reference
- Name the constraint
- Try to change the length of the line using the mouse freezes FreeCAD
  
If the constraint is unnamed there is no problem
Tried to recreate this in the "official" 0.19 AppImage and doesn't crashes
I run the following version:
  

> OS: Fedora 32 (Workstation Edition) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.17.23076 +3024 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 560eb2c4645cd548b6f718c645c7af70384ff54c
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2020-12-22:
I can't reproduce it. How do you change the length using mouse? By dragging the line?

Also, did it freeze or crash? If it crashes, please start the AppImage using command line in a terminal, and post the call stack dump when it crashes? If it freezes, please find out the running FreeCADLink process ID. You can use the task manager on your system. And then kill the process by typing the following command in a terminal, and post the dump here.

```shell
kill -SIGSEGV <ProcessID>
```
```

papamat, 2020-12-22:
Sorry for the wrong term, it freezes FreeCAD it doesn't crashing it.
Trying to recreate today spotted a very weird behavior, so first is the gif that shows how is happening,  
[image]  
  
And next is the same procedure with the only difference being that I press OK button without marking the constraint as reference and then reopen it again and making it reference. This doesn't freezes the app.  
[image]

  
I killed FreeCAD with -SIGSEGV but I have no idea where core is dumped (is not on the working directory), I can imagine that AppImages are dropping dumps in their "virtual space" or something? This is the console output
`/tmp/.mount_FreeCAUsUmMq/AppRun: line 32: 360732 Bus error               (core dumped) "$PREFIX/bin/$name"Link "$@"
`

Shkolik, 2021-07-13:
Tested on:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

No issue found.

luzpaz, 2021-07-13:
Closing ticket

closed 2021-07-13 by luzpaz